realsenseai/librealsense #10600: KinFu constant reset

Amayess, 2022-06-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win (10) |                               |
| Platform                                 | PC |
| SDK Version                            |  2.50.0.0  | 
| Language                            |  {C++ }                          | 
| Segment			|  Robot             | 

### Issue Description

Hi, am trying to compile the **rs-kinfu.cpp** but the program fails to capture pointclouds due to **if (!kf->update(frame))** always staying false.

Is there an explanation for this problem ? 

[image]

MartyG-RealSense, 2022-06-16:
Hi @Amayess  The first step in diagnosing problems with rs-kinfu is always to comment out the line of rs-kinfu.cpp that loads the **High Accuracy** camera configuration preset.  Does the reset error still occur if this line is commented out, please?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp#L204

Amayess, 2022-06-16:
Okay, is there any alternative to the Kinfu algorithm ? after some research i found two alternatives Open3D and INFINITAM.

Which one is the best alternative to the kinfu ? 

ps : my goal is to make a continous scan with the D455 and generate a solide mesh at the end.

MartyG-RealSense, 2022-06-16:
As you are using Windows, if you are able to use commercial software then you could look at the **DotProduct Dot3D** RealSense-compatible 3D modelling software that is D455 compatible and can scan a room.

https://www.dotproduct3d.com/dot3d.html

MartyG-RealSense, 2022-06-24:
HI @Amayess  Do you require further assistance with this case, please?  Thanks!

Amayess, 2022-06-24:
Hi @MartyG-RealSense, no thank you ! I am actually exploring other solutions like Infinitam and Open3D as @dorodnic mentionned it here https://github.com/IntelRealSense/librealsense/issues/6418#issuecomment-630690427

MartyG-RealSense, 2022-06-24:
Thanks very much for the update!  As you do not require further assistance, I will close the case.  Thanks again!

closed 2022-06-24 by MartyG-RealSense