realsenseai/librealsense #10601: How can I get the object distance at, and co-ordinates of, a point on the depth_frame, where the nearest object is the closest to the camera?

yashrivastava, 2022-06-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (05.13.00.50) | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.50.0.3785? }                          | 
| Language                            |  opencv in python                          | 
| Segment			|  others                           | 

### Issue Description

I am essentially trying to find out the pixel on the depth frame that represents the closest point of the nearest object (assuming that it is an irregular shaped object), and also calculate the distance from the camera to the object's closest point. I have managed to output a stream of distances of objects from the camera at a specific hard-coded point on the colour frame. But now I essentially want this point to vary, depending on where the object is nearest to the camera. So far I have not been able to find a solution to this issue. Any help will be much appreciated :).

MartyG-RealSense, 2022-06-17:
Hi @yashrivastava  Some approaches for finding the closest point in a range of depth values with Python are suggested at https://github.com/IntelRealSense/librealsense/issues/3207

yashrivastava, 2022-06-22:
Thank you for the response!

closed 2022-06-22 by yashrivastava