realsenseai/librealsense #10602: npm install node-librealsense =>  npm ERR! Invalid Version: 0.1.0a2

cfry, 2022-06-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  
| Firmware Version                   | I don't actually have a camera yet| 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | MacBook Pro |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | JavaScript                        | 
| Segment			|  {Robot }                          | 

### Issue Description
Installed version of node is 18.0.0 but I also hope to install this in 17.9.0.
I'm trying to use RealSense in a large project that, for other reasons, requires Node 17.9.0
I have no idea what package/software "Version: 0.1.0a2  " refers to.
Looking at the npm log file is not informative to me but I've included it just in case
it makes (real) sense to someone else. Thanks for any help!
[2022-06-17T01_26_35_766Z-debug-0.log](https://github.com/IntelRealSense/librealsense/files/8923884/2022-06-17T01_26_35_766Z-debug-0.log)

MartyG-RealSense, 2022-06-17:
Hi @cfry  My understanding from researching this issue is that this error with 0.1.0a2 can be dealt with by putting the flag **--no-audit** at the start of the npm instruction.  For example:

**npm --no-audit install --save node-librealsense**

The npm link below states that a security audit check that leads to the error occurs automatically when using npm install.  The no-audit flag presumably prevents this error-causing check from occurring.

https://docs.npmjs.com/auditing-package-dependencies-for-security-vulnerabilities

cfry, 2022-06-17:
Thanks Marty.
Your fix did indeed get me further in install but then it broke
on not having the proper C compiler.
That means, I presume that every OS I want to run my code on has to
be compiled for that OS, making release process hard.

Browsers and Node.js now support WebAssembly.
C and C++ code can be transpiled into WebAssembly
(via the Enscripten tool)
which, presumably means that realsense software could be 
converted to WebAssembly and distributed it such that
users wouldn't have to invoke a C complier which is usually problematic.

Has anyone made a Webassembly version of the realsense software?
I don't think this is trivial but its a lot less programmer hours than
each JS user trying to fight the C compiler battles.
I would hope that Intel has, or will soon, do this.

MartyG-RealSense, 2022-06-18:
There is a discussion about installing Node.js on MacOS at https://github.com/IntelRealSense/librealsense/issues/9747

The RealSense user who created that discussion eventually developed an installation guide and shared it at https://github.com/IntelRealSense/librealsense/issues/9747#issuecomment-916925978

abhisheksaini63, 2022-06-20:
I faced this issue in my project and I followed these steps to remove issue 

First you delete node modules 
then delete all unnecessary dependencies from your package.json 
then do npm install 
I thik that is the solution

MartyG-RealSense, 2022-06-20:
Thanks very much @abhisheksaini63 for your advice to @cfry. :)

closed 2022-07-02 by MartyG-RealSense