realsenseai/librealsense #10604: Depth frames not being received, sensor working fine otherwise

anav42, 2022-06-19:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0265326|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003b53046f2|
|**Firmware Update Id**|0003b53046f2|

Errors:
notifications.cpp:511 - Receiver light saturation, stream stopped for 1 sec
notifications.cpp:511 - Error that may be overcome in few sec. 
Stream stoped. May be recoverable

I am using a new to me L515 and not getting any depth frames. I have tried both Linux and Windows as well as the Viewer and Depth Quality tool with no luck. The errors repeat over and over again. I am not sure how to further troubleshoot. Any ideas?

MartyG-RealSense, 2022-06-19:
Hi @anav42  With this particular error, the recommendation is that the camera be returned to the place that it was purchased from.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/10460

Before returning the camera though, please first try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

anav42, 2022-06-19:
Thanks for the reply @MartyG-RealSense . I purchased this used, so I likely can't return it and am looking for a way to repair or address the issue instead.

I am using a USB cable (rated for Thunderbolt 3 and 40 Mbps) with a USB-C connector on both ends. When I flip the orientation of the one attached to the camera, it interestingly doesn't recognize that the camera is even plugged into the computer any more.

I looked at some of the issues you linked (thanks again for that) and it seems like there might be an issue with a thermal measurement or component that relies on that measurement. I took this screenshot which shows an initial error and then about 6 minutes later, multiple, repeating errors. I have not seen the temperature ranges go wonky yet, but before I activated any of the sensors, APD was at -9999 (current status shown in the image below).

Are there any other topics or ideas that you can point me to? Thanks!
[image]

MartyG-RealSense, 2022-06-19:
Use of either USB C to C or Thunderbolt 3 - see case https://github.com/IntelRealSense/librealsense/issues/9445 - increases the chances of problems occurring.

Research of this error in previous cases strongly suggests that it is unrecoverable.  As it is a saturation related error, you could though try reducing the value of the **Laser Power** setting, as increasing laser power can lead to saturation.

MartyG-RealSense, 2022-06-26:
Hi @anav42  Do you require further assistance with this case, please?  Thanks!

anav42, 2022-06-26:
I had ordered a new USB-C to A cable to test out if the issue persists. It does continue to persist and the new cable (while rated for 3.1 and 10Gbps) shows a connection to the computer as a 2.1 speed. Additionally, lowering the Laser Power or any other settings does not seem to produce any depth frames.

Are there any other user level steps I can take? Or is there a service center I can send the unit into? Or at the very least, can additional documentation be provided so that I can perform component level repair on the device?

MartyG-RealSense, 2022-06-26:
There are not any user-repairable parts in the L515 camera model, unfortunately.  And the **Receiver light saturation, stream stopped for 1 sec** error and the inability to access USB 3 mode indicate that there are unresolvable hardware issues with your camera.  

I will discuss your situation with other RealSense support team members.  Thanks very much for your patience in the meantime!

MartyG-RealSense, 2022-06-29:
I have been informed that it is not possible to provide a replacement camera because it was purchased used.

closed 2022-07-22 by MartyG-RealSense