realsenseai/librealsense #10608: Texture count threshold and texture difference threshold

JeffR1992, 2022-06-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win 11   | 
| Kernel Version (Linux Only)    |  (N/A)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0                          | 
| Language                            |  N/A                           | 
| Segment			|  N/A                         | 

### Issue Description
In Advanced Controls -> Depth Control, we have a few interesting parameters. In particular, we have texture count threshold (TCT) and texture difference threshold (TDT).

If TCT is set to 0, then any value given to TDT will have no effect on the depth data. However, for the high density, high accuracy and medium density presets the TCT is set to 0 in each case, yet the TDT is set to different values (1722 for high accuracy, 2466 for high density, and 0 for medium density).

So my question is, since TCT = 0, which disables TDT, why are different values being given to TDT under the high accuracy, high density and medium density presets? I would imagine these different values would have no effect on the data. Any help would be appreciated, thanks.

Also, what does the timeline look like for releasing documentation for your Advanced Controls options?

MartyG-RealSense, 2022-06-20:
Hi @JeffR1992  The Advanced Mode controls interact with each other in complex ways, and so they are undocumented as Intel chose to control them automatically with machine-learning algorithms instead.  There are no future plans to create Advanced Mode documentation.  RealSense users are welcome to experiment with the settings and see how they affect the output though.

There is no information available about the texture count threshold and texture difference threshold functions, other than a couple of non-RealSense references about the TDT technique having been used to measure groove widths in skin.

JeffR1992, 2022-06-23:
Although the advanced mode controls interact in complex ways, each one must have some well defined and logical reason for being used. I understand that using a global optimizer is a great way to deal with tuning a large parameter space, but it seems like a bit of a disservice for Intel to expose so many parameters, yet tell end users to figure it out on their own. Are there at least any well written papers that could give end users more information on what each parameter does instead of having us "shoot in the dark"?  Thanks.

MartyG-RealSense, 2022-06-23:
Point 3 in the 'Play with depth setting' section of Intel's **Tuning depth cameras for best performance** white paper document reiterates the earlier statement that there are not plans to provide future descriptions for what each Advanced Mode parameter does.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#play-with-depth-setting

If you require information about the Advanced Mode features that you enquired about in this case then I am happy to query the Intel RealSense development team about the TCT and TDT functions specifically.

JeffR1992, 2022-06-26:
Hi @MartyG-RealSense , if you could ask the dev team about the issue I mentioned above regarding TCT and TDT that would be appreciated. As far as I can tell, TDT should be dependent on TCT, so I find it interesting that there are presets with TCT = 0 and TDT > 0. It would be good to get a clear definition of how they interact with one another (or better yet a paper defining TCT and TDT via equations).

MartyG-RealSense, 2022-06-26:
I have made the request for the information about the TCT and TDT functions.  Thanks very much for your patience!

MartyG-RealSense, 2022-07-13:
An Intel RealSense colleague has responded with a detailed explanation of the TCT and TDT functions specifically.

*************

-  As you correctly pointed out, TCT is 0 for all of the existing presets, and when TCT=0, TDT is effectively disabled.  Regarding why TDT is >0 for some of these presets, these non-zero values may be a legacy or artifact of the optimization process that was used to create them.  They may have been simply left at these effectively arbitrary values during the tuning process.  Or perhaps they do have a very small effect that is practically negligible.  This is speculation though.

-  My colleague has personally seen no discernible or measurable effect on depth images of changing TDT when TCT=0, which is the case for all of the standard presets and which is consistent with the above assumption.

-  For cases where TCT>0, then increasing TDT will have a noticeable effect.  Basically, more depth will be removed (filtered out) of the images as TDT increases.  The criteria used for this filtering is undocumented but it is not simply a uniform increase in correlation or confidence thresholds like some of the other more basic parameters (neighbor, median, 2nd peak). 
 
It appears to be more selective and based on spatial frequency properties of the image texture.  It tends to have a “high pass” character and so will tend to enhance edges or high contrast regions.  Increasing either TCT or TDT for a given TCT value will tend to sharpen the edges further.

-  As a practical suggestion, you can try increasing TCT (& TDT) for a specific preset and observe the effect.  Lower contrast / smoother regions will begin to be removed.  Whether this is of any value depends on the application but could be useful for highlighting edges of objects and suppressing uniform regions.

**D435 using default preset**

[image]

**Same camera / preset / scene but with TCT / TDT increased to enhance edges.  Note also that the projector is turned Off and RSM bypass is turned On**

[image]

MartyG-RealSense, 2022-07-19:
Hi @JeffR1992  Did the above information answer your questions about TCT and TDT please?  Thanks!

closed 2022-08-01 by MartyG-RealSense