realsenseai/librealsense #10609: Width calculation bugg

D0lan1, 2022-06-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
In the example2 to calculate the height I want to edit it so it shows the width instead. I tried changing  ys = obj_points[:, 1] to ys = obj_points[:, 0] to get width but it shows completely wrong measurements. Will I need to change the np.delete aswell? Since at least the way I'm thinking just changing the object pointer would be enough.

MartyG-RealSense, 2022-06-20:
It looks as though the **np.delete** instruction excludes the background of the image to help to prevent the person being measured in the foreground being confused with z-distance coordinates in the background behind the person.  Changing the value of ys may therefore be disrupting the background removal calculations and providing incorrect measurements.  It may therefore be worth commenting out the np.delete line and having the person that you are measuring stand with their back against a wall to minimize the chance of z-distance affecting the result.

D0lan1, 2022-06-20:
> 

I tried doing it with and without the np.delete and posed myself in front of a wall. The readings were about the same only a difference of 0,1 cm only tested once so might actually just be the same. But I still think I need the np.delete since I will still get outliers from  the detection.  Could it be that the part that does the min max needs to be changed?
`my = np.amin(ys[np.nonzero(ys)], initial=1)
 My = np.amax(ys, initial=-1)
 height = (My - my)` 
 At least the way I'm thinking is that the max values on the x axis will be the most to the right and the min vice versa.

MartyG-RealSense, 2022-06-20:
Instead of changing ys, what happens if the XYZ order that the **verts** are retrieved in is changed from reshape(-1, W, 3) to **reshape(W, -1, 3)**

`verts = np.asanyarray(points.get_vertices()).view(np.float32).reshape(W, -1, 3)  # xyz`

D0lan1, 2022-06-21:
> 

That did not seem to work aswell, noticed that it showed around 1m width even when I stretched my arms out increasing the width of the box.

MartyG-RealSense, 2022-06-27:
Hi @D0lan1  Have you made any progress in the past week with your problem, please?  Thanks!

D0lan1, 2022-06-27:
I've been working on other parts of my project so I've postpone solving the bugg.

MartyG-RealSense, 2022-08-24:
Hi @D0lan1  Do you plan to return to working on the width calculation bug in this case or can the issue be closed, please?  Thanks!

MartyG-RealSense, 2022-09-04:
Case closed due to no further comments received.  If you return to the problem at a future date then please feel free to re-open this issue.  Thanks!

closed 2022-09-04 by MartyG-RealSense