realsenseai/librealsense #10610: No device connected!

kongxin22, 2022-06-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Linux (Ubuntu18)  | 
| Kernel Version (Linux Only)    | Linux 5.4.0-91-generic x86_6４ | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python  | 
| Segment			| Robot        | 

### Issue Description
Hello!When I tried to start the pipeline,I got this error.

Traceback (most recent call last):
  File "/home/kongxin/mmdetection/SOLO/work_dirs/solov2_light_512_dcn_release_r50_fpn_8gpu_1x/realsense/realsensepc.py", line 8, in <module>
    pipeline.start(config)
RuntimeError: No device connected

Process finished with exit code 1

My program was very simple.

import cv2
import pyrealsense2 as rs
import numpy as np

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
pipeline.start(config)(the eighth line)

And I also consulted previous issues.The camera ran well when I used realsense-viewer, and it was not stored in cold and dry conditions.What should I do?Thanks a lot!
[image]

MartyG-RealSense, 2022-06-20:
Hi @kongxin22  Please first test whether starting the pipeline without **config** makes a difference.  With the brackets empty, the program shoud try to run the program using the camera's default stream configuration.

`pipeline.start()`

kongxin22, 2022-06-22:
Thanks for your reply!I followed your suggestion.However,there was no difference.
[image]
[image]
But the camera ran well when I used realsense-viewer.Does it mean my camera is broken?
@MartyG-RealSense

MartyG-RealSense, 2022-06-22:
If the camera works normally when used with the RealSense Viewer then it is not a problem with the camera hardware, the USB port / cable or the camera firmware driver.  The problem is likely to be somewhere in the Python pyrealsense2 wrapper.

I note from your first image that you are using Anaconda instead of Python 3.  This can sometimes cause problems with pyrealsense2.  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/6922#issuecomment-663976346 who used Anaconda with the wrapper and also experienced **RuntimeError: no device connected** shared the solution that worked for them in their particular case.

MartyG-RealSense, 2022-06-22:
Thee are some pyrealsense2 installations where it is necessary to use the following import line where pyrealsense2 is stated twice in order for the script to work.

**import pyrealsense2.pyrealsense2 as rs.**

If that doesn't work and you have switched to using Python 3.6, and you are using a PC as your computer, then you could try installing pyrealsense2 from packages instead of source code with **pip install** if you are able to use pip commands.

`pip install pyrealsense2`

Spj-Zhao, 2022-06-23:
> If the camera works normally when used with the RealSense Viewer then it is not a problem with the camera hardware, the USB port / cable or the camera firmware driver. The problem is likely to be somewhere in the Python pyrealsense2 wrapper.
> 
> I note from your first image that you are using Anaconda instead of Python 3. This can sometimes cause problems with pyrealsense2. A RealSense user at [#6922 (comment)](https://github.com/IntelRealSense/librealsense/issues/6922#issuecomment-663976346) who used Anaconda with the wrapper and also experienced **RuntimeError: no device connected** shared the solution that worked for them in their particular case.

After I don't use anaconda, it runs successfully, thank you very much

kongxin22, 2022-06-23:
I still need to use anaconda for my whole program,though I changed the python version of environment with anaconda.The #6922 issue used anaconda successfully,but I could not install the wrapper from the source as he said.I have uploaded the picture of my installation progress.So could you be so kind to look at it?
[image]

MartyG-RealSense, 2022-06-23:
@kongxin22  I am not familiar with Conda unfortunately.  In regard to pybind11, Conda advice about linking to it is given at https://github.com/IntelRealSense/librealsense/issues/1657#issuecomment-386538660

If you are experiencing **No device connected** in both Conda and Python 3 though then this could suggest that the problem is not related specifically to Conda.

When building for Python 3, does it still not detect the camera if you add **-DPYTHON_EXECUTABLE=/usr/bin/python3 ** to your CMake build instruction?

`cmake ../  -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3`

My personal preference is to build the librealsense SDK and the Python wrapper at the same time instead of building the Python wrapper after librealsense installation has been completed.

`cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

kongxin22, 2022-06-24:
Sorry to bother you.
I found the python wrapper already in the librealsense folder when I installed the SDK. Do you mean I should build the SDK and the Python wrapper separately in a python program? But where is the CMakeLists of it?Due to my failure of installation from the source,can you be so kind to explain the process in detail? 
@MartyG-RealSense

MartyG-RealSense, 2022-06-24:
The code for wrappers such as Python are included in the **wrappers** source code folder of librealsense but are not yet built and so cannot run.  The wrapper needs to be built.

To build the Python wrapper separately, I will assume that you have already downloaded the librealsense source code folder, created a 'build' directory and built the SDK from it.

1.  Go to the SDK's **root** directory, which is the one with a list of folders and files like those in the image below.

[image]

2.  Once in the root folder, input the command **cd build** to navigate to the 'build' folder.

3.  Now input the following sequence of commands.

**cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true
make -j4
sudo make install**

I believe that the build process would make use of the CMakeLists.txt file that is located in the SDK folder's **wrappers/python** directory.

kongxin22, 2022-06-24:
It's so nice of you! I have built it successfully! Thanks for your patience!
At last,does pyrealsense2 have to be built in the environment of Pyhon3.6?

MartyG-RealSense, 2022-06-24:
If the wrapper is built from source code like you have just done then you can use a modern Python 3 version such as 3.10.  After inputting the **sudo make install** instruction, you can then input the command below so that the Python wrapper can find the Python 3 version that you have currently got installed. 

**export PYTHONPATH=$PYTHONPATH:/usr/local/lib**

If it is built from pip packages with **pip install pyrealsense2** then Python 3.9 is the most recent supported version for that installation method at the time of writing this.  The pip install method also does not work on computing devices that use Arm processors, such as Nvidia Jetson and Raspberry Pi.

kongxin22, 2022-06-24:
Thanks for your explanation!
After building successfully,I had a new question!
[image]
Why the pictures are filled with dots?

There is my codes.
[image]

MartyG-RealSense, 2022-06-24:
If there are dots on the color image then it is likely because the infrared dot pattern from the camera's projector is leaking though onto the color image, which it is not meant to do.  It does happen on rare occasions though.

A solution is to reduce the value of the **Laser Power** setting to reduce the visibility of the dots or to turn the projector off completely to make the pattern stop being projected.  Whilst reducing Laser Power or disabling the projector can reduce the quality of the depth image and reduce the brightness of the infrared image, if you are only using the color stream in this script then the color image should not be affected when the pattern is removed.

Below is an example of disabling the projector with Python code.

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
pipeline_profile = pipeline.start(config)
device = pipeline_profile.get_device()
depth_sensor = device.query_sensors()[0]
if depth_sensor.supports(rs.option.emitter_enabled):
depth_sensor.set_option(rs.option.emitter_enabled, 0)
```

kongxin22, 2022-06-24:
I'm so grateful to you. My problems were completely solved.
May you be happy today and always!
@MartyG-RealSense

MartyG-RealSense, 2022-06-24:
Thank you for your kind words.  You are very welcome.   :)

closed 2022-06-30 by MartyG-RealSense