realsenseai/librealsense #10611: RGB color not showing on rosbag point cloud replay

hunterlineage21, 2022-06-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | 01.05.08.01| 
| Operating System & Version |   Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |     5.13.0-48-generic                                  | 
| Platform                                 | PC |
| SDK Version                            |                            | 
| Language                            |  python                      | 
| Segment			| robot                        | 

### Issue Description
I used the realsense-viewer to record and save a point cloud bag file to replay using this [Python script](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py). I was able to play the bag recording but the RGB color does not show; reds are being shown as blue. When I used the same Python script to view a live point cloud, the RGB color shows correctly. I have attached two images, one with the replayed bag file and one with the live point cloud script. How can I fix this issue so that I am viewing the RGB colors correctly even in the replayed bag recording point cloud? Please help urgently

[image]

[image]

MartyG-RealSense, 2022-06-21:
Hi @hunterlineage21  If colours are reversed on the depth image in a program that uses OpenCV instructions then it is usually a problem related to the OpenCV **color space**, a solution for which is described at https://github.com/IntelRealSense/librealsense/issues/9304

If cfg / config is used in a script - as it is in opencv_pointcloud_viewer.py - then converting a program to use a bag file as the data source instead of a live camera usually just requires inserting an **enable_device_from_file** instruction directly before the pipeline start line.  An example of this is provided by the SDK's **read_bag_example.py** Python program.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py#L42

hunterlineage21, 2022-06-21:
@MartyG-RealSense thank you very much for your reply. I was able to use "color_image = cv2.cvtColor(color_image, cv2.COLOR_RGB2BGR)" to get the right colors. I was also wondering how I would be able to make only a specific color of points in the point cloud, say red, to show? I am trying to use "v, t" to do this; what are verts (xyz) and texcoords (uv) defined as?

MartyG-RealSense, 2022-06-21:
As you are using OpenCV, using its **inRange** instruction to perform 'segmentation' removal of colors from the image may meet your needs.  Examples of tutorials that use inRange are below.

https://realpython.com/python-opencv-color-spaces/

https://docs.opencv.org/3.4/da/d97/tutorial_threshold_inRange.html

You can find further examples by googling for the term **opencv python color segmentation inrange**

hunterlineage21, 2022-06-22:
@MartyG-RealSense I am still unsure how I would use cv2.inRange to make only points of a certain color or distance from the origin appear on my point cloud window. I looked at the links to mentioned above but the documentation [here](https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras) explains that the point cloud data obtained is in the form of vertices (x,y,z) and texture-mapping (u,v).

Suppose I have a combined array of vertices and texture-mapping coordinates (say N x 5 array where N are the number of points in the point cloud) how would I use the texture-mapping coordinates to sort out color using BGR values? I understand that we have XYZ available as vertices but how do I perform a query on the texture-mapping UV data to remove a point with a color that I choose?

MartyG-RealSense, 2022-06-22:
If you are using Python and wanted to make only points of a certain distance from the origin appear on the point cloud then you could apply a **threshold filter** post-processing filter to the depth in real-time to set a minimum and maximum distance range for the depth points that are rendered on the image, excluding the depth values outside of that defined range.  There are examples of using the threshold filter with Python at the links below.

https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301

https://www.programmersought.com/article/11995485679/

In regard to filtering color from the RGB image, the tutorial at the link below may be easier to implement as it allows you tp specify the range of color shades that you want to keep (such as between dark blue and light blue) so that when the instruction is applied to an image with blue and also other colors such as red, the non-blue colors will be removed from the image.  This would make it straightforward to define one particular color to keep and remove the rest.

https://www.geeksforgeeks.org/filter-color-with-opencv/

closed 2022-06-29 by hunterlineage21