realsenseai/librealsense #10613: D435 samples run error！labview

SCAROFDEMON, 2022-06-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.05.00  /  05.12.14.50  /  05.13.00.50 | 
| Operating System & Version |    Win10    | 
| Kernel Version (Linux Only)    |  |
| Platform                                 |   PC  |
| SDK Version                           |     v2.33.1   |
| Language                               |   labview    | 
| Segment			         |  just for test    | 

### Issue Description
<Opening RealSense Hello World: Color + Depth，Error 1097 occurs in libRealSense3.lvlib:rs3 open multiple sensor.vi->libRealSense3.lvlib:rs3
maybe：
LabVIEW: (Hex 0x449) An exception occurred in an external node called by a calling library function node. Accidents can cause LabVIEW memory corruption. Save your work in the new location and restart LabVIEW. start all 2.vi->RealSense_HelloWorld_Depth_and_Color.vi call library function node>

I can make <
[image]
>(RealSense_HelloWorld_Depth_and_Monochrome.vi)run with the versions i described  ↑.However，this one （RealSense_HelloWorld_Depth_and_Color.vi）didn't work even i have tried a lot.

MartyG-RealSense, 2022-06-21:
Hi @SCAROFDEMON  Is the **Hello World: Color + Depth** example able to work if you change the **Color Presets** setting?

[image]

Also, if you are using RealSense SDK version 2.33.1 then the recommended firmware to use with that SDK would be **5.1.3.0**.  

The firmwares 5.12.14.50 and 5.13.0.50 are designed for use with SDK version 2.39.0 or newer.

SCAROFDEMON, 2022-06-22:
or you may did the test about “which combination of the firmware and sdk versions runs well with labview“ ？and which labview version will be better one for D435 ？

MartyG-RealSense, 2022-06-22:
A RealSense team member gave detailed advce at https://github.com/IntelRealSense/librealsense/issues/3223 about problems with using the Color + Depth example on D435.  They recommended setting the color to **RGB Ext** to use the RGB sensor of the D435 for color, as the D435 model is not capable of streaming RGB from the left infrared sensor like the D415 can.

In that particular case, SDK 2.18.0 and firmware 5.11.1.0 were being used.  Considering the February 2019 date on that case, they were likely using the May 2018 version of the wrapper as the latest one was not released until 2020.

MartyG-RealSense, 2022-06-28:
Hi @SCAROFDEMON  Do you require further assistance with this case, please?  Thanks!

SCAROFDEMON, 2022-06-28:
Now one of the samples runs well，maybe it's enough for me. you can close the ticket，thanks for your help！

MartyG-RealSense, 2022-06-28:
Thanks very much @SCAROFDEMON for the update!  As you suggested, I will close the ticket.  Thanks again!

closed 2022-06-28 by MartyG-RealSense