realsenseai/librealsense #10614: Can't compile the "Realsense recorder example" from Open3D

Amayess, 2022-06-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Operating System & Version |   Win10  |                                   | 
| Platform                                 | PC.  |
| SDK Version                            |  2.50.0 }                          | 
| Language                            |  C++    | 
| Segment			|  Robot | 

### Issue Description

Hi, is the Intel RealSense D455 compatible with Open3D Lib ? because i tried to compile this example : https://github.com/isl-org/Open3D/blob/master/examples/cpp/RealSenseRecorder.cpp 

But it looks like the program doesn't recognize the D455 realsense since the program doesn't retrieve and display anything at all.. 

Is there any explanation for this issue ? thank you 

[image]
[image]

MartyG-RealSense, 2022-06-21:
Hi @Amayess  What are the resolution and FPS settings that you are using for the depth and color streams, please?

If depth and color were being set to 6 FPS then this would not work on D455 as that camera model uses 5 FPS as its minimum for depth and color instead.

You could also test for the possibility that the camera connection was being identified as being USB 2 instead of USB 3 by setting depth and color to 640x480 and 30 FPS.  Some resolution and FPS combinations are not supported on a USB 2 connection, but 640x480 at 30 FPS is.

Amayess, 2022-06-21:
Thanks for you answer @MartyG-RealSense, after commenting on these lines of code, the program seems to work correctly.

Do the results make sense? especially the intrinsic parameters.

thanks. 

[image]
[image]

closed 2022-06-22 by Amayess