realsenseai/librealsense #10616: Building process of Intel Realsense Legacy v1.12.1 library stops with errors

rodolaf, 2022-06-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200  |  | 
| Operating System & Version | Linux Mint (Ubuntu 14/16/17) | 
| Kernel Version (Linux Only)    | 5.4.0-117 | 
| Platform                                 | PC  |
| SDK Version                            |  Legacy v1.12.1   | 

### Issue Description
Hi, I have tried to build the library in my Linux Mint distribution and it was not possible because the `#include <functional>` was not included in `librealsense/src/types.h` (**maybe a corrupted library??**). After adding this the building process was started but stopped again showing the errors below. Is it again a bug or did I forget to install/configure something?. Thanks in advance

```
/home/xxx/Software/librealsense/src/uvc-v4l2.cpp: In constructor ‘rsimpl::uvc::subdevice::subdevice(const string&)’:
/home/xxx/Software/librealsense/src/uvc-v4l2.cpp:109:66: error: ‘major’ was not declared in this scope
  109 |                 std::ostringstream ss; ss << "/sys/dev/char/" << major(st.st_rdev) << ":" << minor(st.st_rdev) << "/device/";
      |                                                                  ^~~~~
/home/xxx/Software/librealsense/src/uvc-v4l2.cpp:109:94: error: ‘minor’ was not declared in this scope; did you mean ‘mincore’?
  109 |                 std::ostringstream ss; ss << "/sys/dev/char/" << major(st.st_rdev) << ":" << minor(st.st_rdev) << "/device/";
      |                                                                                              ^~~~~
      |                                                                                              mincore
make[2]: *** [CMakeFiles/realsense.dir/build.make:323: CMakeFiles/realsense.dir/src/uvc-v4l2.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:189: CMakeFiles/realsense.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

MartyG-RealSense, 2022-06-22:
Hi @rodolaf  I am pleased to hear that you achieved a solution.  I will refer your question about the source files to the RealSense team.

MartyG-RealSense, 2022-06-23:
Hi again @rodolaf  Intel informed me that there is a newer version of Legacy Librealsense (1.12.4) than the 1.12.1 version you are using where the issue that you referred to is fixed. 

https://github.com/IntelRealSense/librealsense/tree/v1.12.4

rodolaf, 2022-06-23:
Hi Marty @MartyG-RealSense , thanks a lot for clariying this. Just as a hint, it would probably great if also the link to the legacy release in the main Intel Realsense page *https://github.com/IntelRealSense/librealsense) points also to the latest release. Then you can avoid some questions like the ones I have placed

MartyG-RealSense, 2022-06-23:
Thanks very much @rodolaf - I have passed this suggestion to Intel.

rodolaf, 2022-06-23:
Hi @MartyG-RealSense, unfortunately the Release v1.12.4 wasn´t much better.... When I execute `make && sudo make install` after following all previous steps, I get the following errors, and this time, I have run out of ideas on how to solve it :(
`
/home/xxx/Software/librealsense/build/third-party/libusb/libusb/core.c:23:10: fatal error: config.h: No such file or directory
   23 | #include "config.h"
      |          ^~~~~~~~~~
compilation terminated.
make[5]: *** [CMakeFiles/usb.dir/build.make:76: CMakeFiles/usb.dir/libusb/core.c.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:83: CMakeFiles/usb.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/libusb.dir/build.make:87: libusb-prefix/src/libusb-stamp/libusb-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:833: CMakeFiles/libusb.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
`

MartyG-RealSense, 2022-06-23:
I will refer your latest question to Intel.  Thanks very much.

hsuys, 2022-06-23:
@rodolaf, can you give the following steps a try, please?
1. Install libusb-1.0 
`sudo apt-get install libusb-1.0-0-dev`
3. Remove/cleanup previous build folder then re-run cmake and make.

rodolaf, 2022-06-23:
Hi @hsuys , thanks for reaching me out. Your suggestion helped a bit but after a while a new error was shown : (

`...
[ 52%] Building CXX object examples/software-device/CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o

In file included from /home/rodolaf/Software/librealsense/examples/software-device/../example.hpp:10,
                 from /home/rodolaf/Software/librealsense/examples/software-device/rs-software-device.cpp:6:
/usr/include/GLFW/glfw3.h:210:13: fatal error: GL/glu.h: No such file or directory
  210 |    #include <GL/glu.h>
      |             ^~~~~~~~~~
compilation terminated.
make[2]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/build.make:76: examples/software-device/CMakeFiles/rs-software-device.dir/rs-software-device.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1046: examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
`

hsuys, 2022-06-24:
@rodolaf, it seems like the glfw3 library is missing in order to build the examples codes. 
Please kindly refer to [Linux Installation ](https://github.com/IntelRealSense/librealsense/blob/v1.12.4/doc/installation.md#3rd-party-dependencies).

rodolaf, 2022-06-24:
... and I get new errors. After executing `./scripts/patch-realsense-ubuntu-xenial.sh`and hitting "Enter" hundreds of times to answer a lot of annoying questions that I do not know how to answer, I get the following errors (remember that now I have a Linux Mint distribution

`/bin/bash: ./scripts/ubuntu-retpoline-extract-one: No such file or directory
make[2]: *** [scripts/Makefile.build:289: arch/x86/purgatory/purgatory.o] Error 127
make[2]: *** Deleting file 'arch/x86/purgatory/purgatory.o'
make[1]: *** [arch/x86/Makefile:243: archprepare] Error 2
make: *** [Makefile:525: __build_one_by_one] Error 2`

rodolaf, 2022-06-26:
Hi @hsuys, I have managed to see the streams on screen launching for example _"cpp-capture"_ BUT after a while it shows the error _"rs.warn: Subdevice 0 bad magic number 0x0"_. I know the reason for that is an unpatched uvcvideo driver ([181](https://github.com/IntelRealSense/librealsense/issues/181)), so no need to discuss this again here. Running examples without uvcvideo (only the terminal) like _"cpp-tutorial-1-depth"_ works totally fine with my R200.

For other users with similar problems, if I am honest, I do not really know how I've got this running but it started working after executing in the terminal `sudo modprobe uvcvideo`, like [this user](https://github.com/IntelRealSense/librealsense/issues/7055) did. Do not ask me for more details because I cannot give you  : )

@hsuys , you can close this issue  since I got now what I was looking for. I know it is not fun to give support on old stuff but I really appreciate dedicating a bit of your time on this. @MartyG-RealSense , also many thanks for your support

closed 2022-06-26 by rodolaf

MartyG-RealSense, 2022-06-26:
@rodolaf  It's great to hear that you were successful.  :)

@hsuys. Thanks very much for the support provided to @rodolaf