realsenseai/librealsense #10617: Run rs-software-device examples only get black board.  Empty point cloud on rs2::pointcloud::calculate()

linyx21, 2022-06-22:
I build project on windows 10 ,Visual Studio 2019. when run rs-software-device example only get black board without text "This point-cloud is generated from a synthetic device:".   like the issue  [Empty point cloud on rs2::pointcloud::calculate()](https://github.com/IntelRealSense/librealsense/issues/5728)

MartyG-RealSense, 2022-06-22:
Hi @linyx21  I tested the pre-built .exe version of **rs-software-device** in the Tools folder of the RealSense SDK's Windows version (2.50.0) and confirmed the black screen and text message.  I will refer the issue to the RealSense team.  Thanks for your patience!

[image]

MartyG-RealSense, 2022-06-22:
An Intel bug report has now been created for this issue.  I have added an **Enhancement** label to this case to signify that it should be kept open whilst the bug report is active, but you do not need to take further action.  Thanks again for reporting the issue!

linyx21, 2022-06-23:
Thanks! It's running well.

MartyG-RealSense, 2022-06-23:
This case needs to remain open as there is now an Intel bug report associated with it.  You do not need to do anything else though if you are satisfied.  Thanks for the update!

MartyG-RealSense, 2022-08-24:
The software_device black screen issue has had a fix implemented in the new **2.51.0** version of the SDK.  I will therefore close this case.  Thanks again!

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

commit 8a1f951c (2022-07-17): PR #10619 from Guoping: fix #10617 rs-software-device black output
closed 2022-08-24 by MartyG-RealSense