realsenseai/librealsense #10620: converting bag files to image frames

Pshcer, 2022-06-23:
Hi 
I am using windows 11.
I have captured a bag file using real sense D415 and now I want to convert it to picture frames.
Is there any step-by-step procedure to accomplish this task?

I am a new user.

MartyG-RealSense, 2022-06-23:
Hi @Pshcer  The RealSense SDK has a tool called **rs-convert** that can convert frames from a bag file to a range of formats including PNG images.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

If you have installed the full RealSense SDK on Windows using the **Intel.RealSense.SDK-WIN10** installer file in the 'Assets' file list of the SDK **Releases** page then a pre-built version of the tool called **rs-convert.exe** that can be launched with a double-leftclick on the file can be found at the folder location below:

C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools

If the full RealSense SDK has not been installed yet on your computer then you can find its installer file at the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

Pshcer, 2022-06-23:
@MartyG-RealSense Thank you for your feedback,
I have installed the same package and the rs-converter is there but I can't open it by double-clicking it.
I have tried the code suggested in the link but an error is coming
[image]

MartyG-RealSense, 2022-06-23:
My apologies, I was thinking of another example in that folder that can be double-leftclicked and have edited my comment above to correct it.  This example program needs to be run from the **Command Prompt** interface of Windows.  You can find the Command Prompt by typing **command** into the search box at the bottom of the Windows screen, next to the corner button where the computer is shut down.

In regard to the error in the code version that you tried, it looks as though you have not put the .bag extension on the end of the bag filename so the program cannot find it.

**20220622_154037.bag**

Pshcer, 2022-06-23:
@MartyG-RealSense 
Thanks, the commands are working now , but i cant find the converted file in the paths provided
[image]

the images are processed but unable to locate its location(not there in the path provided)

MartyG-RealSense, 2022-06-23:
Please try removing the " " quotation marks from around the C:/ path.

MartyG-RealSense, 2022-06-23:
That's great to hear.  You are very welcome!

closed 2022-06-24 by Pshcer