realsenseai/librealsense #10621: Extract the pixel data of each frame

Amayess, 2022-06-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Win 10  |                                   
| Platform                                 | PC  |
| SDK Version                            |  2.50.0.0 } | 
| Language                            |  C++       | 
| Segment			|  Robot| 

### Issue Description
Hi @MartyG-RealSense, I am looking for a way to extract the pixel coordinates (x,y,z) of the first 40 frames, so far now i have been able to extract the pixel data using the RealSense example "Rs-hello" that i modified like this : 
[image]

Is this possible to have an access to the whole data information of each pixel for each frame ? Thank you

MartyG-RealSense, 2022-06-23:
Hi @Amayess  You can obtain the XYZ of a specific pixel by converting a color pixel to a depth pixel with **rs2_project_color_pixel_to_depth_pixel**.  C++ examples can be found at https://github.com/IntelRealSense/librealsense/issues/2982 and https://github.com/IntelRealSense/librealsense/issues/5728

If you wish to be able to access the XYZ of all coordinates then generating a point cloud and storing its **vertices** with the **points.get_vertices()** instruction may a helpful approach, as described at https://github.com/IntelRealSense/librealsense/issues/1783

Amayess, 2022-06-23:
> Hi @Amayess You can obtain the XYZ of a specific pixel by converting a color pixel to a depth pixel with **rs2_project_color_pixel_to_depth_pixel**. A C++ example can be found at #2982
> That's what i found too, thank you ! 
> If you wish to be able to access the XYZ of all coordinates then generating a point cloud and storing its **vertices** with the **points.get_vertices()** instruction to retrieve XYZ from the vertices may a helpful approach, as described at #1783 and #5728

What if i want to get the distance between every points or pixels contained in a depth frame ?

MartyG-RealSense, 2022-06-23:
The SDK has a C++ example program called **rs-measure** that can measure the real-world distance between two points.  So conceivably you could find the distance between any two points from coordinate A to coordinate B by changing the values of A and B to the coordinates that you want to compare the distance between.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

MartyG-RealSense, 2022-06-30:
Hi @Amayess  Do you require further assistance with this case, please?  Thanks!

Amayess, 2022-06-30:
Hi @MartyG-RealSense, I am still working on it.

Can I close it when i will finish ? Thank you

MartyG-RealSense, 2022-06-30:
Yes, it is fine to close it when you are ready to do so.  Thanks very much for the update!