realsenseai/librealsense #10623: converting bag files to image frames in ubuntu

rohan3301, 2022-06-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 

| Operating System & Version |  Linux (Ubuntu 20.04) | 

### Issue Description
I have captured the bag file using Realsense 415 and want to covert in into image frames.
is there any step by step procedure to accomplish this task.

I am a beginner in this .

MartyG-RealSense, 2022-06-24:
Hi @rohan3301  There is a pre-made tool called **rs-convert** that can extract frames from a bag file and convert them into a range of file formats, including PNG images.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

If you have built the librealsense SDK on Ubuntu 20.04 with examples and tools included then there should be a pre-built executable version of the tool in the **usr/local/bin** folder.

rohan3301, 2022-06-24:
@MartyG-RealSense
 I have installed the realsense viewer.exe file alone.
where can i have the complete package file

MartyG-RealSense, 2022-06-24:
There are a range of ways to install the full librealsense SDK on Ubuntu, such as building it from Debian packages or from source code.  

**Debian packages**
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

**Source code**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

************

If you have a Windows computer available to you then you can also access a pre-built exe of the rs-convert tool by downloading and running the **Intel.RealSense.SDK-WIN10** installer from the 'Assets' file list of the SDK **Releases** page.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

On Windows this example program can be run from the **Command Prompt** interface of Windows.  You can find the Command Prompt by typing command into the search box at the bottom of the Windows screen, next to the corner button where the computer is shut down.

Once the Command Prompt window is launched, you can navigate to the folder where the rs-convert program is located by inputting this command:

**cd C:\Program Files (x86)\Intel RealSense SDK 2.0\tools**

rohan3301, 2022-06-24:
Thank you worked.

closed 2022-06-24 by rohan3301