realsenseai/librealsense #10624: nodelet manager process has died, exit code -11, dmesg --> usb usb2-port3: cannot disable, segfault

byhang, 2022-06-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.13.0-36-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.45.0 }                          | 
| Language                            |  c++ }                          | 
| Segment			|  Robot }                          | 

### Issue Description
Hi Intel,
I got error when launch official realsense-ros-2.3.0, use my launch file which copy from rs_camera.launch and include/nodelet.launch.xml.

# roslaunch error:
[ WARN] [1656065577.814077970]: Param '/fl_cam/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 200000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1656065577.827148874]: Param '/fl_cam/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1656065577.962120107]: Done Setting Dynamic reconfig parameters.
[ INFO] [1656065577.963537216]: depth stream is enabled - width: 848, height: 480, fps: 15, Format: Z16
[ INFO] [1656065577.964731016]: color stream is enabled - width: 640, height: 480, fps: 6, Format: RGB8
[ INFO] [1656065577.968293423]: setupPublishers...
[fl_cam/fl_cam_manager-2] process has died [pid 8828, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=fl_cam_manager __log:=/home/byhang/.ros/log/2a5b1e34-f3a5-11ec-86dd-74e5f9ec52bb/fl_cam-fl_cam_manager-2.log].
log file: /home/byhang/.ros/log/2a5b1e34-f3a5-11ec-86dd-74e5f9ec52bb/fl_cam-fl_cam_manager-2*.log

# dmesg error
[   82.557328] audit: type=1400 audit(1656065110.496:34): apparmor="DENIED" operation="open" profile="/usr/lib/telepathy/mission-control-5" name="/etc/dconf/profile/user" pid=2878 comm="mission-control" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[  113.897267] usb 2-2: USB disconnect, device number 2
[  116.283063] usb 2-2: new SuperSpeed USB device number 4 using xhci_hcd
[  116.303670] usb 2-2: New USB device found, idVendor=8086, idProduct=0b07
[  116.303677] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  116.303682] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435 
[  116.303686] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[  116.303690] usb 2-2: SerialNumber: 901523021387
[  116.305049] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  116.306993] uvcvideo: Unable to create debugfs 2-4 directory.
[  116.307146] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  116.307153] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
[  116.307158] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
[  116.307485] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input21
[  116.308545] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  116.309485] uvcvideo: Unable to create debugfs 2-4 directory.
[  116.309666] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
[  116.309672] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
[  116.309677] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
[  117.522720] xhci_hcd 0000:00:14.0: Cannot set link state.
[  117.522732] usb usb2-port3: cannot disable (err = -32)
[  117.522743] usb 2-3: USB disconnect, device number 3
[  117.523917] hid-sensor-hub 0003:8086:0B3A.0001: No report with id 0xffffffff found
[  117.523939] hid-sensor-hub 0003:8086:0B3A.0001: No report with id 0xffffffff found
[  117.524161] hid-sensor-hub 0003:8086:0B3A.0001: No report with id 0xffffffff found
[  119.082831] usb 2-3: new SuperSpeed USB device number 5 using xhci_hcd
[  119.103201] usb 2-3: New USB device found, idVendor=8086, idProduct=0b3a
[  119.103202] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  119.103203] usb 2-3: Product: Intel(R) RealSense(TM) Depth Camera 435i
[  119.103204] usb 2-3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[  119.103205] usb 2-3: SerialNumber: 903223053633
[  119.104302] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  119.105887] uvcvideo: Unable to create debugfs 2-5 directory.
[... 226 lines cut ...]
      <param name="topic_odom_in"            type="str"  value="$(arg topic_odom_in)"/>
      <param name="calib_odom_file"          type="str"    value="$(arg calib_odom_file)"/>
      <param name="publish_odom_tf"          type="bool" value="$(arg publish_odom_tf)"/>
      <param name="stereo_module/exposure/1" type="int"  value="$(arg stereo_module/exposure/1)"/>
      <param name="stereo_module/gain/1"     type="int"  value="$(arg stereo_module/gain/1)"/>
      <param name="stereo_module/exposure/2" type="int"  value="$(arg stereo_module/exposure/2)"/>
      <param name="stereo_module/gain/2"     type="int"  value="$(arg stereo_module/gain/2)"/>

      <param name="filters"                  type="str"    value="$(arg filters)"/>
      <param name="clip_distance"            type="double" value="$(arg clip_distance)"/>
      <param name="linear_accel_cov"         type="double" value="$(arg linear_accel_cov)"/>
      <param name="initial_reset"            type="bool"   value="$(arg initial_reset)"/>
      <param name="reconnect_timeout"        type="double" value="$(arg reconnect_timeout)"/>
      <param name="wait_for_device_timeout"  type="double" value="$(arg wait_for_device_timeout)"/>
      <param name="unite_imu_method"         type="str"    value="$(arg unite_imu_method)"/>
    </node>
    <node pkg="nodelet" type="nodelet" name="depth_points"
        args="load depth_image_proc/point_cloud_xyz $(arg manager) --no-bond"
        respawn="$(arg respawn)" output="$(arg output)">
      <remap from="image_rect" to="/$(arg camera)/aligned_depth_to_color/image_raw"/> 
      <remap from="points"     to="/$(arg camera)/aligned_depth_to_color/points" />
    </node> 
    <node pkg="topic_tools" type="throttle" name="$(arg camera)_color_hz_control" args="messages /$(arg camera)/color/image_raw 6 /$(arg camera)/color/image_raw/fix_rate" output="$(arg output)"/>
  </group>
</launch>

MartyG-RealSense, 2022-06-24:
Hi @byhang  Is launch able to complete successfully if you use the original rs_camera.launch instead of your own copy of it?

Also, I note that your roslaunch log is identifying two separate cameras with unique serial numbers, a D435 and D435i.  Can you confirm whether you are launching more than one camera, please?  Thanks!

[image]

byhang, 2022-06-26:
Hi @MartyG-RealSense 
We use D435 D435i and D430, randomly, multi camera. 
I use the original rs_camera.launch, got error, too.

➜  ~ roslaunch realsense2_camera rs_camera.launch camera:=fcb_hu_cam usb_port_id:=2-2             
... logging to /home/byhang/.ros/log/2a5b1e34-f3a5-11ec-86dd-74e5f9ec52bb/roslaunch-Eros-28314.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Eros:41353/

SUMMARY
========

PARAMETERS
 * /fcb_hu_cam/realsense2_camera/accel_fps: -1
 * /fcb_hu_cam/realsense2_camera/accel_frame_id: fcb_hu_cam_accel_...
 * /fcb_hu_cam/realsense2_camera/accel_optical_frame_id: fcb_hu_cam_accel_...
 * /fcb_hu_cam/realsense2_camera/align_depth: False
 * /fcb_hu_cam/realsense2_camera/aligned_depth_to_color_frame_id: fcb_hu_cam_aligne...
 * /fcb_hu_cam/realsense2_camera/aligned_depth_to_fisheye1_frame_id: fcb_hu_cam_aligne...
 * /fcb_hu_cam/realsense2_camera/aligned_depth_to_fisheye2_frame_id: fcb_hu_cam_aligne...
 * /fcb_hu_cam/realsense2_camera/aligned_depth_to_fisheye_frame_id: fcb_hu_cam_aligne...
 * /fcb_hu_cam/realsense2_camera/aligned_depth_to_infra1_frame_id: fcb_hu_cam_aligne...
 * /fcb_hu_cam/realsense2_camera/aligned_depth_to_infra2_frame_id: fcb_hu_cam_aligne...
 * /fcb_hu_cam/realsense2_camera/allow_no_texture_points: False
 * /fcb_hu_cam/realsense2_camera/base_frame_id: fcb_hu_cam_link
 * /fcb_hu_cam/realsense2_camera/calib_odom_file: 
 * /fcb_hu_cam/realsense2_camera/clip_distance: -2.0
 * /fcb_hu_cam/realsense2_camera/color_fps: -1
 * /fcb_hu_cam/realsense2_camera/color_frame_id: fcb_hu_cam_color_...
 * /fcb_hu_cam/realsense2_camera/color_height: -1
 * /fcb_hu_cam/realsense2_camera/color_optical_frame_id: fcb_hu_cam_color_...
 * /fcb_hu_cam/realsense2_camera/color_width: -1
 * /fcb_hu_cam/realsense2_camera/confidence_fps: -1
 * /fcb_hu_cam/realsense2_camera/confidence_height: -1
 * /fcb_hu_cam/realsense2_camera/confidence_width: -1
 * /fcb_hu_cam/realsense2_camera/depth_fps: -1
 * /fcb_hu_cam/realsense2_camera/depth_frame_id: fcb_hu_cam_depth_...
 * /fcb_hu_cam/realsense2_camera/depth_height: -1
 * /fcb_hu_cam/realsense2_camera/depth_optical_frame_id: fcb_hu_cam_depth_...
 * /fcb_hu_cam/realsense2_camera/depth_width: -1
 * /fcb_hu_cam/realsense2_camera/device_type: 
 * /fcb_hu_cam/realsense2_camera/enable_accel: False
 * /fcb_hu_cam/realsense2_camera/enable_color: True
 * /fcb_hu_cam/realsense2_camera/enable_confidence: True
 * /fcb_hu_cam/realsense2_camera/enable_depth: True
 * /fcb_hu_cam/realsense2_camera/enable_fisheye1: False
 * /fcb_hu_cam/realsense2_camera/enable_fisheye2: False
 * /fcb_hu_cam/realsense2_camera/enable_fisheye: False
 * /fcb_hu_cam/realsense2_camera/enable_gyro: False
 * /fcb_hu_cam/realsense2_camera/enable_infra1: False
 * /fcb_hu_cam/realsense2_camera/enable_infra2: False
 * /fcb_hu_cam/realsense2_camera/enable_infra: False
 * /fcb_hu_cam/realsense2_camera/enable_pointcloud: False
 * /fcb_hu_cam/realsense2_camera/enable_pose: False
 * /fcb_hu_cam/realsense2_camera/enable_sync: False
 * /fcb_hu_cam/realsense2_camera/filters: 
 * /fcb_hu_cam/realsense2_camera/fisheye1_frame_id: fcb_hu_cam_fishey...
 * /fcb_hu_cam/realsense2_camera/fisheye1_optical_frame_id: fcb_hu_cam_fishey...
 * /fcb_hu_cam/realsense2_camera/fisheye2_frame_id: fcb_hu_cam_fishey...
 * /fcb_hu_cam/realsense2_camera/fisheye2_optical_frame_id: fcb_hu_cam_fishey...
 * /fcb_hu_cam/realsense2_camera/fisheye_fps: -1
 * /fcb_hu_cam/realsense2_camera/fisheye_frame_id: fcb_hu_cam_fishey...
 * /fcb_hu_cam/realsense2_camera/fisheye_height: -1
 * /fcb_hu_cam/realsense2_camera/fisheye_optical_frame_id: fcb_hu_cam_fishey...
 * /fcb_hu_cam/realsense2_camera/fisheye_width: -1
 * /fcb_hu_cam/realsense2_camera/gyro_fps: -1
 * /fcb_hu_cam/realsense2_camera/gyro_frame_id: fcb_hu_cam_gyro_f...
[... 90 lines cut ...]
[ 2666.522239] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
[ 2666.522242] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
[ 2714.803929] usb 2-2: USB disconnect, device number 11
[ 2718.639787] usb 2-2: new SuperSpeed USB device number 12 using xhci_hcd
[ 2718.660401] usb 2-2: New USB device found, idVendor=8086, idProduct=0b07
[ 2718.660408] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2718.660412] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435 
[ 2718.660417] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[ 2718.660420] usb 2-2: SerialNumber: 901523021387
[ 2718.661760] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[ 2718.663733] uvcvideo: Unable to create debugfs 2-12 directory.
[ 2718.663943] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 2718.663948] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
[ 2718.663952] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
[ 2718.664199] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input29
[ 2718.664971] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[ 2718.665836] uvcvideo: Unable to create debugfs 2-12 directory.
[ 2718.666027] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
[ 2718.666032] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
[... cut ...] era 6 was not initialized!
[ 2955.783495] nodelet[28398]: segfault at 1 ip 00007f692c3394a1 sp 00007f6903703a80 error 6 in librostime.so[7f692c322000+2b000]
[ 3226.320203] hid-sensor-hub 0003:8086:0B3A.0006: No report with id 0xffffffff found
[ 3226.357131] hid-sensor-hub 0003:8086:0B3A.0006: No report with id 0xffffffff found
[ 3230.068119] hid-sensor-hub 0003:8086:0B3A.0006: No report with id 0xffffffff found
[ 3230.068405] hid-sensor-hub 0003:8086:0B3A.0006: No report with id 0xffffffff found

MartyG-RealSense, 2022-06-26:
The cause of the 'entity type' kernel messages tends to be difficult to trace and if they are corrected then you may never find out for sure what caused them in the first place.  I previously enquired to the RealSense development team about them in 2020 and received the following explanation.

*********

The message “Entity camera ... not initialized" is related to UVC topology defined in the firmware descriptor for USB device. 
 Those are protocol-specific messages and are apparently stemmed from firmware.

**********

The most recent RealSense ROS user who experienced the errors suspected at https://github.com/IntelRealSense/librealsense/issues/10158#issuecomment-1028817808 that the problem had occurred because they had several different versions of the librealsense SDK installed on their computer. Is the presence of multiple installations a possibility on your own computer, please?

byhang, 2022-06-26:
Hi @MartyG-RealSense , 
Thanks for your quick reply.

This is my installed deb version, all are the default, lasted versions in Ubuntu-16.04.
librealsense2_2.50.0-0~realsense0.6126_amd64.deb
librealsense2-dkms_1.3.18-0ubuntu1_all.deb
librealsense2-gl_2.50.0-0~realsense0.6126_amd64.deb
librealsense2-net_2.50.0-0~realsense0.6126_amd64.deb
librealsense2-udev-rules_2.50.0-0~realsense0.6126_amd64.deb
librealsense2-utils_2.50.0-0~realsense0.6126_amd64.deb
ros-kinetic-ddynamic-reconfigure_0.2.0-0xenial-20210503-114111-0800_amd64.deb
ros-kinetic-librealsense2_2.45.0-1xenial-20210506-081043-0800_amd64.deb
ros-kinetic-realsense2-camera_2.3.0-1xenial-20210506-084151-0800_amd64.deb

> Building for 4.13.0-36-generic and 4.15.0-142-generic
> Building initial module for 4.13.0-36-generic
> Secure Boot not enabled on this system.
Maybe dkms error?

# install log
➜  sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  daemontools fonts-droid-fallback fonts-noto-mono libatk-bridge2.0-dev libatspi2.0-dev libepoxy-dev libglfw3 libgmp-dev libgmpxx4ldbl libgtk-3-dev libwnck-common libwnck22 libxfce4ui-1-0 libxfce4ui-common libxfce4util-bin
  libxfce4util-common libxfce4util7 libxfconf-0-2 libxtst-dev nettle-dev x11proto-record-dev xfconf
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  librealsense2-udev-rules
The following NEW packages will be installed:
  librealsense2-dkms librealsense2-udev-rules
0 upgraded, 2 newly installed, 0 to remove and 887 not upgraded.
Need to get 10.3 MB of archives.
After this operation, 81.7 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://librealsense.intel.com/Debian/apt-repo xenial/main amd64 librealsense2-dkms all 1.3.18-0ubuntu1 [10.3 MB]
Get:2 https://librealsense.intel.com/Debian/apt-repo xenial/main amd64 librealsense2-udev-rules amd64 2.50.0-0~realsense0.6126 [7,788 B]                                                                                              
Fetched 10.3 MB in 6s (1,522 kB/s)                                                                                                                                                                                                    
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 565549 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.18-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.18-0ubuntu1) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.50.0-0~realsense0.6126_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.50.0-0~realsense0.6126) ...
Setting up librealsense2-dkms (1.3.18-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.18 DKMS files...
First Installation: checking all kernels...
Building for 4.13.0-36-generic and 4.15.0-142-generic
Building initial module for 4.13.0-36-generic
Secure Boot not enabled on this system.
Done.

uvcvideo:
Running module version sanity check.

 - Original module
 - Installation
   - Installing to /lib/modules/4.13.0-36-generic/updates/dkms/

videodev.ko:
Running module version sanity check.

 - Original module
 - Installation
   - Installing to /lib/modules/4.13.0-36-generic/updates/dkms/

hid-sensor-accel-3d.ko:
Running module version sanity check.
[... 80 lines cut ...]
The following NEW packages will be installed:
  ros-kinetic-ddynamic-reconfigure ros-kinetic-librealsense2 ros-kinetic-realsense2-camera
0 upgraded, 3 newly installed, 0 to remove and 887 not upgraded.
Need to get 13.1 MB of archives.
After this operation, 36.9 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-librealsense2 amd64 2.45.0-1xenial-20210506-081043-0800 [12.8 MB]
Get:2 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-ddynamic-reconfigure amd64 0.2.0-0xenial-20210503-114111-0800 [57.7 kB]                                                                                        
Get:3 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-realsense2-camera amd64 2.3.0-1xenial-20210506-084151-0800 [308 kB]                                                                                            
Fetched 13.1 MB in 1min 30s (144 kB/s)                                                                                                                                                                                                
Selecting previously unselected package ros-kinetic-librealsense2.
(Reading database ... 571989 files and directories currently installed.)
Preparing to unpack .../ros-kinetic-librealsense2_2.45.0-1xenial-20210506-081043-0800_amd64.deb ...
Unpacking ros-kinetic-librealsense2 (2.45.0-1xenial-20210506-081043-0800) ...
Selecting previously unselected package ros-kinetic-ddynamic-reconfigure.
Preparing to unpack .../ros-kinetic-ddynamic-reconfigure_0.2.0-0xenial-20210503-114111-0800_amd64.deb ...
Unpacking ros-kinetic-ddynamic-reconfigure (0.2.0-0xenial-20210503-114111-0800) ...
Selecting previously unselected package ros-kinetic-realsense2-camera.
Preparing to unpack .../ros-kinetic-realsense2-camera_2.3.0-1xenial-20210506-084151-0800_amd64.deb ...
Unpacking ros-kinetic-realsense2-camera (2.3.0-1xenial-20210506-084151-0800) ...
Processing triggers for libc-bin (2.23-0ubuntu10) ...
Setting up ros-kinetic-librealsense2 (2.45.0-1xenial-20210506-081043-0800) ...
Setting up ros-kinetic-ddynamic-reconfigure (0.2.0-0xenial-20210503-114111-0800) ...
Setting up ros-kinetic-realsense2-camera (2.3.0-1xenial-20210506-084151-0800) ...
Processing triggers for libc-bin (2.23-0ubuntu10) ...

MartyG-RealSense, 2022-06-26:
From the information at the top of this case, it looks as though your intention was to use librealsense 2.45.0 and its matching ROS wrapper 2.3.0.  Indeed, this is shown in the roslaunch log.

[image]

In your latest comment at https://github.com/IntelRealSense/librealsense/issues/10624#issuecomment-1166489616 the logs have references to both librealsense 2.45.0 and 2.50.0, and two different kernel versions (4.13.0-36-generic and 4.15.0-142-generic).  

In that comment, it also logs that the ros wrapper's 'Method 1' installation process of **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** was used.  This would install both the librealsense SDK and the ROS wrapper from packages at the same time.  This install method will also install the currently latest librealsense version and its matching wrapper.  So using that command today would not install 2.45.0 and 2.3.0, but instead install librealsense 2.50.0 and wrapper 2.3.2.

So if you had already installed librealsense 2.45.0 previously and then ran the Method 1 command then you could end up with two potentially conflicting installations of different versions of librealsense on the same computer.

My recommendations would therefore be:

1.  If you have previously installed the ROS wrapper from source code then delete the entire ROS catkin workspace folder **/catkin_ws/src/**

2.  Run the command below to uninstall all librealsense-related packages.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

3.  Install librealsense 2.50.0 and wrapper 2.3.2 together at the same time from packages with the command below.

`sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`

byhang, 2022-06-27:
Hi @MartyG-RealSense 
sudo apt-get install ros-$ROS_DISTRO-realsense2-camera=2.3.2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version '2.3.2' for 'ros-kinetic-realsense2-camera' was not found

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
The newest version is 2.3.0 in my env, How can I do for official version 2.3.2?

# apt search librealsense
Sorting... Done
Full Text Search... Done
librealsense2/xenial,now 2.50.0-0~realsense0.6126 amd64 [installed,automatic]
  Intel(R) RealSense(tm) Cross Platform API - runtime

librealsense2-dbg/xenial 2.50.0-0~realsense0.6126 amd64
  Intel(R) RealSense(tm) Camera Capture API - debug symbols

librealsense2-dev/xenial 2.50.0-0~realsense0.6126 amd64
  Intel(R) RealSense(tm) Camera Capture API - development files

librealsense2-dkms/xenial,now 1.3.18-0ubuntu1 all [installed]
  Modified kernel modules for librealsense2

librealsense2-gl/xenial,now 2.50.0-0~realsense0.6126 amd64 [installed,automatic]
  Intel(R) RealSense(tm) - GLSL-enabled extensions

librealsense2-gl-dbg/xenial 2.50.0-0~realsense0.6126 amd64
  Intel(R) RealSense(tm) GLSL extensions - debug symbols

librealsense2-gl-dev/xenial 2.50.0-0~realsense0.6126 amd64
  Intel(R) RealSense(tm) Camera Capture API - development files

librealsense2-net/xenial,now 2.50.0-0~realsense0.6126 amd64 [installed,automatic]
  Intel(R) RealSense(tm) - Network Bridge extension

librealsense2-net-dbg/xenial 2.50.0-0~realsense0.6126 amd64
  Intel(R) RealSense(tm) Network Bridge extension - debug symbols

librealsense2-net-dev/xenial 2.50.0-0~realsense0.6126 amd64
  Intel(R) RealSense(tm) Camera Capture API - Network Bridge extension development files

librealsense2-udev-rules/xenial,now 2.50.0-0~realsense0.6126 amd64 [installed,automatic]
  Intel(R) RealSense(tm) Camera Capture API - udev rules

librealsense2-utils/xenial,now 2.50.0-0~realsense0.6126 amd64 [installed]
  Intel(R) RealSense(tm) Camera Capture API - utils and demos

realsense-sdk-dev/xenial 2.7.7-0~realsense0.29 amd64
  Intel(R) RealSense(tm) Camera Capture API - development files

realsense-sdk-utils/xenial 2.7.7-0~realsense0.29 amd64
  Intel(R) RealSense(tm) Camera Capture API - utils and demos

ros-kinetic-librealsense/xenial 1.12.2-1xenial-20210506-081035-0800 amd64
  Library for capturing data from the Intel(R) RealSense(TM) F200, SR300, R200, LR200 and ZR300 cameras.

ros-kinetic-librealsense2/xenial,now 2.45.0-1xenial-20210506-081043-0800 amd64 [installed,automatic]
  Library for capturing data from the Intel(R) RealSense(TM) SR300, D400 Depth cameras and T2xx Tracking devices.

byhang, 2022-06-27:
Hi @MartyG-RealSense ,
Are there any bugs in official realsense-ros-2.3.0?

I have to test at 2.3.0, because the newest official version is 2.3.0 in Ubuntu-16.04.
But I got same error, 
use:
>librealsense2-2.45.0
ros-kinetic-librealsense2-2.45.0
ros-kinetic-realsense2-camera-2.3.0
Ubuntu-16.04
4.13.0-36-generic kernel only (4.15.0-* are removed, than reinstall realsense deb)
official rs_camera.launch

### $ dpkg -l | grep realsense
> ii  librealsense2-dkms                           1.3.18-0ubuntu1                                       all          Modified kernel modules for librealsense2
ii  librealsense2-udev-rules:amd64               2.45.0-0~realsense0.4550                              amd64        Intel(R) RealSense(tm) Camera Capture API - udev rules
ii  ros-kinetic-librealsense2                    2.45.0-1xenial-20210506-081043-0800                   amd64        Library for capturing data from the Intel(R) RealSense(TM) SR300, D400 Depth cameras and T2xx Tracking devices.
ii  ros-kinetic-realsense2-camera                2.3.0-1xenial-20210506-084151-0800                    amd64        RealSense Camera package allowing access to Intel T265 Tracking module and SR300 and D400 3D cameras

### $ roslaunch realsense2_camera rs_camera.launch usb_port_id:=2-2
[ INFO] [1656321482.793619608]: Stereo Module was found.
[ INFO] [1656321482.810608329]: RGB Camera was found.
[ INFO] [1656321482.811011033]: Motion Module was found.
[ INFO] [1656321482.811061586]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1656321482.811094068]: num_filters: 0
[ INFO] [1656321482.811118709]: Setting Dynamic reconfig parameters.
[ WARN] [1656321483.256434565]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 200000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1656321483.270204556]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1656321483.398787511]: Done Setting Dynamic reconfig parameters.
[ INFO] [1656321483.400456437]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1656321483.404092036]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1656321483.404170844]: setupPublishers...
[camera/realsense2_camera_manager-1] process has died [pid 7319, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/byhang/.ros/log/8af7c8e0-f5f9-11ec-bba1-74e5f9ec52bb/camera-realsense2_camera_manager-1.log].
log file: /home/byhang/.ros/log/8af7c8e0-f5f9-11ec-bba1-74e5f9ec52bb/camera-realsense2_camera_manager-1*.log

### dmesg
[  268.246154] nodelet[5891]: segfault at 1 ip 00007f762eef24a1 sp 00007f760127ba80 error 6 in librostime.so[7f762eedb000+2b000]

MartyG-RealSense, 2022-06-27:
Regarding the **[camera/realsense2_camera_manager-1] process has died** error at the end of the launch process, a RealSense ROS user of Kinetic experienced it in https://github.com/IntelRealSense/realsense-ros/issues/473 and the solution for them in their particular case was a completely fresh installation.

In another similar case at https://github.com/IntelRealSense/realsense-ros/issues/1774 where the error number was -2 instead of -1, building the ROS wrapper from source code instead of packages resolved it.

byhang, 2022-06-28:
Thank you @MartyG-RealSense very much for your careful assistance.
In Ubuntu-16.04, I build ddynamic-reconfigure-0.3.2、 realsense-ros-2.3.2 and librealsense2-2.50.0 from source code, than I solved this problem.

closed 2022-06-28 by byhang