realsenseai/librealsense #10626: D415  C++/OpenCV Problem Displaying RGB and Depth Images

MoniqueReal, 2022-06-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415
| Firmware Version                   | 05.13.00.50 
| Operating System & Version |   Win (10)  
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | HP-Pavillion i7 Laptop  |
| SDK Version                            | 2.50.0.3785                          | 
| Language                            |  C++/OpenCV/MS Visual Studio 
| Segment			|                           | 

### Issue Description
Regarding the below github C++ program that is documented to display RGB frames And corresponding Depth frames, 
when the program is run both the RGB and Depth frames only display a solid gray color.  I am new to C++, OpenCV, as well as RealSense software.  Any assistance is greatly appreciated!  Monique
 
Original File Location:  https://github.com/Taeyoung96/RealSense-Example/blob/master/realsenseVisualize.cpp

#include <iostream>
#include <librealsense2/rs.hpp>
#include <opencv2/core/core.hpp>
#include <opencv2/opencv.hpp>

using namespace std;
using namespace cv;

// Visualize Depth Image(with out Colorization) and RGB Image using OpenCV

int main() {
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    rs2::config cfg;

    cfg.enable_stream(RS2_STREAM_DEPTH);
    cfg.enable_stream(RS2_STREAM_COLOR);

    // Start streaming with default recommended configuration
    pipe.start();

    // Defbine two align objects. One will be used to align
    // to depth viewport and the other to color.
    // Creating align oject is an expensive operation
    // that should not be performed in the main loop
    rs2::align align_to_color(RS2_STREAM_COLOR);

    // Declare Window name
    const auto window_name = "Display Depth";
    // namedWindow(window_name, CV_WINDOW_AUTOSIZE); replaced with below line
       namedWindow(window_name,    WINDOW_AUTOSIZE);

    while (true)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        auto aligned_frames = align_to_color.process(data); // align to depth viewpoint with color image

        // Get Depth and Color frame
        rs2::depth_frame depth = aligned_frames.get_depth_frame();
        rs2::frame color = aligned_frames.get_color_frame();

        // Declare width and height
        const int w_depth = depth.get_width();
        const int h_depth = depth.get_height();

        const int w_img = color.as<rs2::video_frame>().get_width();
        const int h_img = color.as<rs2::video_frame>().get_height();

        // Create OpenCV matrix of size (w,h) from the colorized depth data
        Mat imageDepth(Size(w_depth, h_depth), CV_16UC1, (void*)depth.get_data(), Mat::AUTO_STEP);
        // Create OpenCV matrix of size (w,h) from the RGB image data
        Mat imageRGB(Size(w_img, h_img), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);

        //cv::cvtColor(imageRGB, imageRGB, CV_BGR2RGB); replaced with below line
        cv::cvtColor(imageRGB, imageRGB, COLOR_BGR2RGB, 0);

        // Update the window with new depth data
        imshow(window_name, imageDepth);
        // Update the window and show RGB Image
        imshow("frame", imageRGB);
    }

    return 0;
}

MartyG-RealSense, 2022-06-26:
Hi @MoniqueReal  This OpenCV script is part of a larger project created by a RealSense user.  

https://github.com/Taeyoung96/RealSense-Example

If you are new to RealSense, OpenCV and C++ then Intel's official OpenCV examples may be a better place to begin.  In particular, the example **rs-grabcuts** demonstrates OpenCV depth / color alignment like in the 'realsenseVisualize.cpp' script that you were using.

**Getting Started**
https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md

**RealSense OpenCV examples page**
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv

**rs-grabcuts - depth and colour alignment**
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/grabcuts

MoniqueReal, 2022-06-26:
Thank you MartyG for your kind detailed reply.  My objective is to design a C++/OpenCV program that outputs simultaneously both continuous RGB and corresponding Depth images so that I may perform real-time operations on the images.  I'll follow through on the examples that you provided to learn how to display these images much like those provided by that stand-alone  Intel RealSense Viewer V2.50.0.  Monique

MartyG-RealSense, 2022-06-26:
You are very welcome.  Please do feel free to ask further questions as you test the examples.  Good luck!

closed 2022-07-18 by MartyG-RealSense