realsenseai/librealsense #10627: 2 net devices (Same or different camera model) can't work in same process.

JunSun-CSH, 2022-06-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D415/D455 } | 
| Operating System & Version |   {Win 10 (Client) / Debian 11 (Server)|                                    | 
| Platform                                 | PC/Raspberry Pi 4B |
| SDK Version                            |  { 2.50 }                          | 
| Language                            |  {C++ }                          | 

### Issue Description
I built a rs-server and run rs-server on a raspberry pi 4B board with Debian 11 to make the D415(D455) work as a net device.

In other Win10 PC, wrote codes to connect D415 and D455 net devices in in 2 separated processes. I was successfully connect to them and can capture the RGB and depth images.

But, if I tried to run these 2 net devices in same process, the second net device can't start normally.

Also tried to connect same model realsense cameras (D455), these net device can be started and work. But, the second net device's intrinsics and extrinsics parameter values were the first net device's values. This lead failed to align depth&RGB .

MartyG-RealSense, 2022-06-27:
Hi @JunSun-CSH  The white-paper guide document for the rs-server ethernet networking system recommends placing each individual camera on its own computing board and not using multiple cameras on the same board.

[image]

**EtherSense**, the Python-based predecessor of the rs-server system, recommended the same 'one camera, one board' approach.

https://github.com/krejov100/EtherSense

[image]

Can you confirm whether you are putting its camera on its own computing board, please?

My understanding from https://github.com/IntelRealSense/librealsense/issues/6376 is that if a network camera script uses **config.enable_device** to identify individual Network Device cameras by their unique serial number then they should be treated the same way by the script as non-networked cameras are.

MartyG-RealSense, 2022-07-04:
Hi @JunSun-CSH  Do you require further assistance with this case, please?  Thanks!

JunSun-CSH, 2022-07-04:
> Hi @JunSun-CSH Do you require further assistance with this case, please? Thanks!

Hi Marty,

Yes, I put one camera on one Raspberry Pi 4B board.

MartyG-RealSense, 2022-07-04:
If your goal is to access multiple network cameras from the same application, an appropriate strategy may be to take the RealSense SDK's **rs-multicam** multiple camera example program and modify it to access the two cameras as network cameras instead.

**rs-multicam**
https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp

**Modifying existing scripts for network cameras**
https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#34-sdk-examples

JunSun-CSH, 2022-07-05:
> If your goal is to access multiple network cameras from the same application, an appropriate strategy may be to take the RealSense SDK's **rs-multicam** C++ multiple camera example program and modify it to access the two cameras as network cameras instead.
> 
> **rs-multicam** https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp
> 
> **Modifying existing scripts for network cameras with C++** https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#34-sdk-examples

Hi Marty,

I tried your recommend way, but this also didn't work and show same behavior: The second camera can't be started.

If use the debug version realsense2d.dll and realsense2-netd.dll to run the test application, I will get option not support error .

MartyG-RealSense, 2022-07-05:
If you are able to run the D415 and D455 in two separate processes simultaneously (but not in the same process) then the problem is not likely to be related to the power supplied to the cameras.  

Could you provide more information please about how you use 'two separate processes'.  Is this two separate shell terminals?

closed 2022-07-17 by MartyG-RealSense