realsenseai/librealsense #10628: Remove the dependecies from OS ???

jiapei-nexera, 2022-06-27:
There seems to be **NO** point to rely on **ROS** for media reading or recording ????

My personal suggestions....

MartyG-RealSense, 2022-06-27:
Hi @jiapei-nexera  If you are referring to the RealSense SDK's use of the **bag** file format for recording and playback of camera data then although the format is based on **rosbag**, it is not necessary to have ROS or the RealSense ROS wrapper installed to record and play the SDK's version of the bag format.

jiapei-nexera, 2022-06-27:
@MartyG-RealSense 

Yes, that is what I wanna express...
But, is there a simple way for me to tick in CMakeLists.txt and have that **rosbag** excluded from building???

MartyG-RealSense, 2022-06-27:
As far as I am aware, the SDK bag format is a core part of the librealsense SDK and not a third-party dependency.  Some RealSense users have chosen to read bag files via the **rosbag API** instead of using the SDK-bag system and doing so may involve installing dependencies.  

http://wiki.ros.org/rosbag/Code%20API

An example of this would be the **RealSense ROS-bag parser** project in the Community Projects section of the SDK's examples page.  The documentation for that project states that "the dependencies are opencv 3.x, glog, librealsense, and the part of the ros library included as headers, and boost".

https://github.com/IntelRealSense/librealsense/tree/master/examples#community-projects

If bags are recorded and played back directly with the SDK though with its built-in script instructions **enable_record_to_file**  and **enable_device_from_file** then those functions should not require special dependencies to be built just for the purpose of handling bags.

If you do not need graphical support at all in your librealsense build - for example, you are using a 'headless' system without a display - then you can exclude graphics support and graphical tools / examples from the build by including the build flag **-DBUILD_GRAPHICAL_EXAMPLES=false** in your CMake build instruction.

jiapei-nexera, 2022-06-27:
To tell the truth, I just wanna remove all third_party libraries, which seems to be **almost there**, but **ONLY except ROSBAG** ...

```console
[ 10%] Building CXX object CMakeFiles/realsense2.dir/src/pipeline/profile.cpp.o
ccache /usr/bin/c++ -DBUILD_SHARED_LIBS -DCOM_MULTITHREADED -DEASYLOGGINGPP_ASYNC -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DHWM_OVER_XU -DRS2_USE_CUDA -DRS2_USE_V4L2_BACKEND -DSQLITE_HAVE_ISNAN -DUNICODE -DUSING_UDEV -Drealsense2_EXPORTS -I/usr/local/cuda/include -I....../librealsense/src -I....../librealsense/include -I/usr/include/libusb-1.0 -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mssse3 -fopenmp -fPIC -std=gnu++11 -MD -MT CMakeFiles/realsense2.dir/src/media/ros/ros_writer.cpp.o -MF CMakeFiles/realsense2.dir/src/media/ros/ros_writer.cpp.o.d -o CMakeFiles/realsense2.dir/src/media/ros/ros_writer.cpp.o -c ....../librealsense/src/media/ros/ros_writer.cpp
ccache /usr/bin/c++ -DBUILD_SHARED_LIBS -DCOM_MULTITHREADED -DEASYLOGGINGPP_ASYNC -DELPP_NO_DEFAULT_LOG_FILE -DELPP_THREAD_SAFE -DHWM_OVER_XU -DRS2_USE_CUDA -DRS2_USE_V4L2_BACKEND -DSQLITE_HAVE_ISNAN -DUNICODE -DUSING_UDEV -Drealsense2_EXPORTS -I/usr/local/cuda/include -I....../librealsense/src -I....../librealsense/include -I/usr/include/libusb-1.0 -pedantic -g -Wno-missing-field-initializers -Wno-switch -Wno-multichar -Wsequence-point -Wformat -Wformat-security -mssse3 -fopenmp -fPIC -std=gnu++11 -MD -MT CMakeFiles/realsense2.dir/src/pipeline/profile.cpp.o -MF CMakeFiles/realsense2.dir/src/pipeline/profile.cpp.o.d -o CMakeFiles/realsense2.dir/src/pipeline/profile.cpp.o -c ....../librealsense/src/pipeline/profile.cpp
In file included from ....../librealsense/src/media/ros/ros_reader.cpp:5:
....../librealsense/src/media/ros/ros_reader.h:6:10: fatal error: rosbag/view.h: No such file or directory
    6 | #include "rosbag/view.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:611: CMakeFiles/realsense2.dir/src/media/ros/ros_reader.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from ....../librealsense/src/media/playback/playback_device.cpp:8:
....../librealsense/src/media/ros/ros_reader.h:6:10: fatal error: rosbag/view.h: No such file or directory
    6 | #include "rosbag/view.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:583: CMakeFiles/realsense2.dir/src/media/playback/playback_device.cpp.o] Error 1
In file included from ....../librealsense/src/pipeline/profile.cpp:6:
....../librealsense/src/media/ros/ros_writer.h:5:10: fatal error: rosbag/bag.h: No such file or directory
    5 | #include "rosbag/bag.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
In file included from ....../librealsense/src/media/playback/playback_sensor.cpp:10:
....../librealsense/src/media/ros/ros_reader.h:6:10: fatal error: rosbag/view.h: No such file or directory
    6 | #include "rosbag/view.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:1143: CMakeFiles/realsense2.dir/src/pipeline/profile.cpp.o] Error 1
make[2]: *** [CMakeFiles/realsense2.dir/build.make:597: CMakeFiles/realsense2.dir/src/media/playback/playback_sensor.cpp.o] Error 1
In file included from ....../librealsense/src/context.cpp:17:
....../librealsense/src/media/ros/ros_reader.h:6:10: fatal error: rosbag/view.h: No such file or directory
    6 | #include "rosbag/view.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:1479: CMakeFiles/realsense2.dir/src/context.cpp.o] Error 1
In file included from ....../librealsense/src/pipeline/pipeline.cpp:8:
....../librealsense/src/media/ros/ros_writer.h:5:10: fatal error: rosbag/bag.h: No such file or directory
    5 | #include "rosbag/bag.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:1115: CMakeFiles/realsense2.dir/src/pipeline/pipeline.cpp.o] Error 1
In file included from ....../librealsense/src/media/ros/ros_writer.cpp:14:
....../librealsense/src/media/ros/ros_writer.h:5:10: fatal error: rosbag/bag.h: No such file or directory
    5 | #include "rosbag/bag.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:625: CMakeFiles/realsense2.dir/src/media/ros/ros_writer.cpp.o] Error 1
In file included from ....../librealsense/src/rs.cpp:15:
....../librealsense/src/media/ros/ros_writer.h:5:10: fatal error: rosbag/bag.h: No such file or directory
    5 | #include "rosbag/bag.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/realsense2.dir/build.make:1675: CMakeFiles/realsense2.dir/src/rs.cpp.o] Error 1
make[2]: Leaving directory '....../librealsense/build'
make[1]: *** [CMakeFiles/Makefile2:122: CMakeFiles/realsense2.dir/all] Error 2
make[1]: Leaving directory '....../librealsense/build'
make: *** [Makefile:139: all] Error 2

```

MartyG-RealSense, 2022-06-27:
There is an SDK third-party folder location called **third-party/realsense-file/ rosbag**.

https://github.com/IntelRealSense/librealsense/tree/master/third-party/realsense-file/rosbag

The original design philosophy of the RealSense SDK 2.0 (librealsense2) at its inception in 2017 was to minimize the number of third-party dependencies so that the SDK can work on as many different hardware and software configurations as possible.  The usual method of excluding certain elements from the build is to set particular build flags to False, as you have done.  I am not aware of previous cases where rosbag support has been disabled though.

There is a Getting Started project where the approach is taken of excluding all dependencies from the **project** instead of the SDK.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

jiapei-nexera, 2022-06-27:
@MartyG-RealSense 

Looks to me: **RosBag** is **ONLY** for compression while writing, and de-compressing while reading ....

jiapei-nexera, 2022-06-27:
@MartyG-RealSense 

Take a look at this file [https://github.com/IntelRealSense/librealsense/blob/master/src/media/playback/playback_device.cpp](https://github.com/IntelRealSense/librealsense/blob/master/src/media/playback/playback_device.cpp), did you notice `#include "media/ros/ros_reader.h"`, and then,  you should be able to trace to [https://github.com/IntelRealSense/librealsense/blob/master/src/media/ros/ros_reader.h](https://github.com/IntelRealSense/librealsense/blob/master/src/media/ros/ros_reader.h)  

Clearly, no matter what you do during configuration, **librealsense** relies on 
```console
#include "rosbag/view.h"
#include "ros_file_format.h"
``` 
In one word, there is **NO** possibility to let **librealsense** independent from **rosbag**, which actually did **NOTHING BUT Compression and De-compression** ...

I've got **NO** idea how **Intel** is collaborating with **Stanford ROS**, but it looks this kind of job does NOT make any sense to **RealSense**... You can use **libpng** or anything else to do you own streaming ....

MartyG-RealSense, 2022-06-27:
Thank you very much for explaining your reasoning.  It helped me to understand.

A RealSense team member explains at https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766 how the RealSense SDK's bag format has some differences from the rosbag format.

jiapei-nexera, 2022-06-27:
@MartyG-RealSense 
I got it ...

**writer** and **reader** from `serialization.h` are abstract classes, they **must** be implemented  by something else, that is why **rosbag** .... hmmmmm'

I personally do NOT like **ROS** ....

MartyG-RealSense, 2022-07-04:
Hi @jiapei-nexera  Do you require further assistance with this case, please?  Thanks!

jiapei-nexera, 2022-07-04:
@MartyG-RealSense 
Nope... I need more help from [https://github.com/isl-org/Open3D](https://github.com/isl-org/Open3D) ...

closed 2022-07-04 by jiapei-nexera