realsenseai/librealsense #10632: Sharing insight on how I got my librealsense SDK to work!

JoshuaSummers, 2022-06-28:
Hey! Since the people here helped me get my project to work, I wanted to make a little tutorial to give back to the community. I had IMMENSE difficulty getting my librealsense SDK to work on my rpi 3, and I wanted to share how I did it. If this isn't the place for this, let me know! I can just upload this to my own github page, haha.
-------------------------------------------------
This is what finally worked for me to be able to use my Intel RealSense D435 Camera with a raspberry pi:

First, I flashed my SD card with a pre-built image. You’ll know if this works if when you boot your raspberry pi up, you have a screensaver of an Intel RealSense camera.
https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras?_ga=
(The step "2.4 Assembly Steps” is not necessary. I stopped doing the tutorial before doing steps “2.7 Testing the Camera” and later)

After typing sudo reboot, in step 2.6, I increased the swap file on my Raspberry Pi. I think this helps with RAM problems. This is a great tutorial: https://pimylifeup.com/raspberry-pi-swap-file/
(I used conf_swapsize=1024, some people input conf_swapsize=2048 instead, it doesn’t matter much.)

From here, I used this tutorial: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

Start at the “install TBB” section and complete that first.

I ignored the “install OpenCV” section, you can just do pip3 install opencv-python later

I completed the rest of the tutorial, but after I input a cmake command and it completed, I typed sudo reboot. After rebooting, I then typed the make -j1 command. Make sure you type make -j1 in the correct directory though after rebooting (type cd ~/librealsense/build first, then make -j1). This helped with downloading speeds. I also let my screen go to sleep while I was waiting for the make -j1 command to process. The screen going to sleep actually made it go way faster.

This whole thing can take a while depending on how much RAM your raspberry pi has. Downloading the librealsense sdk part took 3 hours for me!

If this still doesn’t work for you, I would recommend these two other tutorials that look promising, but I haven’t tried them.

1. https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

2. https://github.com/datasith/Ai_Demos_RPi/issues/8 (Coderooni’s comment)
  
There’s also this thread that looks pretty good: https://support.intelrealsense.com/hc/en-us/community/posts/360048495493--Intel-Realsense-D435-with-RaspberryPi-Best-practice-installation-guide

Best of luck! Getting the sdk part to work was one of the worst parts of my project. Looking up the errors you receive is the best way to fix your problems in my opinion.

MartyG-RealSense, 2022-06-29:
Thanks so much @JoshuaSummers for sharing the experience from your Raspberry Pi 3 setup process with the RealSense community!

JoshuaSummers, 2022-06-29:
> Thanks so much @JoshuaSummers for sharing the experience from your Raspberry Pi 3 setup process with the RealSense community!

Yep! Of course Marty :D

closed 2022-06-29 by JoshuaSummers