realsenseai/librealsense #10633: RealSense release memory

SowisLiuYiming, 2022-06-29:
RealSense release memory

I have a problem now.
use wait_for_frames  or  poll_for_frames  get new frames.
Expected time: 	<2s
Actual time:	5s or timeout

I think the application maintains too long history, the device may lack memory resources to produce new frames, and the following call to this method shall fail to retrieve new frames.
So I want to **free up memory** every once in a while.What should I do?

SowisLiuYiming, 2022-06-29:
<html>
<body>

Required Info |  
-- | --
Camera Model | D415
Firmware Version | 5.12.7.100
Operating System & Version | Ubuntu 16.04
Kernel Version (Linux Only) | 4.15.0-142-generic
SDK Version | 2.38.1
Language | C++

</body>
</html>

MartyG-RealSense, 2022-06-29:
Hi @SowisLiuYiming  You can release frames with the instruction **rs2_release_frame()**.  A RealSense team member provides C++ advice about use of this instruction at https://github.com/IntelRealSense/librealsense/issues/4006

SowisLiuYiming, 2022-06-30:
Hi @MartyG-RealSense.
Thanks for your reply, I will try this method.

SowisLiuYiming, 2022-07-01:
Hi @MartyG-RealSense 
My environment：Multi-camera（5 sensor），USB3 hub，bandwidth（1.43Gbps）Lost frames appear in continuous shooting.pipeline.wait_for_frames()，This method will execute for 2 seconds or more.

I try this method.Its arguments don't match my current code.rs2_release_frame(**rs2_frame* frame**) 。I didn't use **rs2_frame** in my code.Here's my code. How do I solve this problem. 

rs2::frameset frameset;
rs2::frame unFilter;
rs2::frame filtered;
rs2::pipeline pipeline;

frameset = pipeline.wait_for_frames();
unFilter = frameset.get_depth_frame();
filtered = unFilter;
filtered = temp_filter.process(filtered);

auto vf = frameset.get_color_frame().as<rs2::video_frame>()
auto vf2= filtered.as<rs2::video_frame>();
auto vf3 = frameset.get_depth_frame().as<rs2::video_frame>();

rs2_release_frame（？？？）

MartyG-RealSense, 2022-07-01:
Hi @SowisLiuYiming  If you are using 5 cameras attached to the same computer, are they all enabled simultaneously please?  If they are then there is the possibility that the computer's processor is being over-burdened and causing slowdown.  The more cameras that are attached to the same computer, the greater the amount of the computer's resources that will be consumed.  In the past Intel have recommended using a computer with at least an Intel i7 CPU for 4 cameras.

You could use an Ubuntu system monitoring tool such as **htop** to check whether the percentage usage of the computer's CPU is near or at 100% when 5 cameras are active simultaneously.

If the 5 cameras are being accessed by the same program (a **multicam** application) then Intel also recommend using the **poll_for_frames()** instruction instead of wait_for_frames(), as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

closed 2022-07-13 by MartyG-RealSense