realsenseai/librealsense #10635: module 'pyrealsense2' has no attribute 'pipeline' and 'context'

jiapei-nexera, 2022-06-29:
A couple of issues:

1. For Python - pyrealsense2
- manually installed pyrealsense2 always meet the above error messages:
module 'pyrealsense2' has no attribute 'pipeline' and 'context'

- directly `pip install` failed (and surely not preferred)
```console
✗ pip install -U --user pyrealsense2
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
```

2. For C++

- `rs-hello-realsense.cpp` runs successfully, but how to view?
-  `realsense-viewer` is either ugly, or the project management is ugly.  from the top-level CMakeLists.txt, 

`include(common/utilities/CMakeLists.txt)` , `common/utilities is built`, but all the rest under `common` are **NOT**?

jiapei-nexera, 2022-06-29:
Done, by copy that `__init__.py` ...

closed 2022-06-29 by jiapei-nexera

Nir-Az, 2024-06-26:
Should be fixed now with the following PR #13079