realsenseai/librealsense #10637: How to align depth camera to my custom mobile phone RGB camera

myalos, 2022-06-30:
I want to align depth photo to my custom mobile phone RGB.  How to do that？

MartyG-RealSense, 2022-06-30:
Hi @myalos   Is your phone an Android device, please?  If it is then you could align depth to an RGB image from the RealSense camera instead of the mobile phone's RGB using the RealSense Android wrapper.  An example of doing so is at https://github.com/IntelRealSense/librealsense/issues/4716

If you need to align to a mobile phone RGB sensor and you are able to use C++ code then using OpenCV with the RealSense SDK's **software_device** system would be a recommendable route to take to achieve your goal.  An example of this is at https://github.com/IntelRealSense/librealsense/issues/10067

More information about software_device can be found at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

MartyG-RealSense, 2022-07-08:
Hi @myalos  Do you require further assistance with this case, please?  Thanks!

myalos, 2022-07-08:
Thanks for quick reply !  I don't need further assistance.

MartyG-RealSense, 2022-07-08:
Thanks very much @myalos for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-07-08 by MartyG-RealSense