realsenseai/librealsense #10640: Record a number of frames then export it to a single PLY file

Amayess, 2022-07-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.13.00.50 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC |
| SDK Version                            |2.50.0.0  | 
| Language                            |  C++   | 
| Segment			|  Robot      | 

### Issue Description

Hi,

In order to generate a complete mesh without gaps ( (because RealSense only takes the last frame to make the mesh), I was wondering if it was possible that in a first time I record let's say 5 frames containing a point cloud in a continuous way and then record the content of these five in the same PLY file? I have already tried with similar algorithms like KinFu or Open3D but without success.

I saw that it's possible to use the affine transform method to superimpose and align several point clouds, but I don't know if it is suitable for what I want to do or not?

Thank you.

MartyG-RealSense, 2022-07-01:
Hi @Amayess  If your project is able to make use of an application outside of the RealSense SDK then you could investigate the possibility of saving 5 individual ply files in the SDK and then merging them together in the open-source RealSense-compatible C++ tool **CloudCompare**.  This tool is available for Windows, Linux and Mac.

**CloudCompare**
https://cloudcompare.org/

https://github.com/cloudcompare/cloudcompare

**Cloudcompare point cloud merging**
http://www.cloudcompare.org/doc/wiki/index.php?title=Merge

http://www.cloudcompare.org/forum/viewtopic.php?t=2543

Amayess, 2022-07-04:
Hi @MartyG-RealSense, 

Thank you for your suggestion but actually I am looking for a way to do it without using an application outside of the RealSenseSDK. 

I was thinking about doing a loop to generate let's say 5 frames then save them in a single PLY file using the affine transform method to superimpose each PLY.

Is this possible ? 

My goal is to generate a complete mesh with no gaps.

Thank you.

MartyG-RealSense, 2022-07-04:
I researched your request carefully but was unable to find an example of code that could perform the described method with a single camera within the RealSense SDK only.

The SDK does have an official C++ example program called **rs-pointcloud-stitching** for stitching pointclouds from two cameras into a single image.  This program makes use of the affine transform instruction **rs2_transform_point_to_point**

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pointcloud/pointcloud-stitching

Amayess, 2022-07-04:
Hi @MartyG-RealSense thank you for your reactivity ! 

Okay understood, what about the fact **generate a specfic number of frames** using a loop ? 

Logically, it's possible,  no ?

MartyG-RealSense, 2022-07-04:
If you use a **for int** instruction then the script will be forced to loop a command for a specified number of frames before the script is permitted to move onwards past the 'for' instruction and carry out the rest of the script.  For example, a for-int loop to capture 30 frames exactly:

```
rs2::frameset frames;
for(int i = 0; i < 30; i++)
{
//Wait for all configured streams to produce a frame
frames = pipe.wait_for_frames();
}
```

In this example, the program should only be permitted to capture 30 frames before the pipe.wait_for_frames() instruction can no longer be initialized.

So conceivably you could change '30' to '5' to capture 5 frames only and place between { and } some code to generate a point cloud frame.  The instruction would loop for 5 frames before the loop ends.  

You could then place pointcloud stitching code after the closing } bracket so that it only stitches the frames once 5 of them have been generated and the loop ends, permitting the script to move onwards to the next line outside of the for-int closing bracket.

However, if auto-exposure is enabled then the first several frames after the pipeline starts may be bad because auto-exposure requires time to settle down.  In the SDK's **rs-save-to-disk** C++ example, it deals with this by using a first for-int loop to wait for 30 frames, and then has a second for-int loop to do processing work on the 31st frame onwards now that auto-exposure has settled.

https://github.com/IntelRealSense/librealsense/blob/master/examples/save-to-disk/rs-save-to-disk.cpp#L29-L34

closed 2022-07-04 by Amayess