realsenseai/librealsense #10641: how to calibrate lidar l515 and how to use pointcloud in python

saisatheesh5, 2022-07-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (Open RealSense Viewer --> 2.50 | 
| Operating System & Version |   {Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)   | (5.4)                                     | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hai, I am using L515 lidar camera. In my case, I need to measure the distance between two points in an object and I need to measure the angle of the bending object. suggest some links to solve my issue. i think I want to use the point cloud to detect the particular pixel point is correct? and how to calibarate my camera to measure the accurate result.

MartyG-RealSense, 2022-07-01:
Hi @saisatheesh5  A Python script for measuring the distance between two points is provided at the link below.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

For measuring the angle of a bending object, a discussion at https://github.com/IntelRealSense/librealsense/issues/9297 about measuring the angle of an object's height from the ground may be a helpful reference.

In regard to calibration, the L515 camera model does not have a calibration tool for its depth and RGB sensors.

saisatheesh5, 2022-07-02:
Thank you so much MartyG. How do I get a particular pixel value in an object automatically using the point cloud?

saisatheesh5, 2022-07-02:
def calculate_distance(self,x,y):
        color_intrin = self.color_intrin
        ix,iy = self.ix, self.iy
        udist = self.depth_frame.get_distance(ix,iy)
        vdist = self.depth_frame.get_distance(x, y)
        #print udist,vdist

        point1 = rs.rs2_deproject_pixel_to_point(color_intrin, [ix, iy], udist)
        point2 = rs.rs2_deproject_pixel_to_point(color_intrin, [x, y], vdist)
        #print str(point1)+str(point2)

        dist = math.sqrt(
            math.pow(point1[0] - point2[0], 2) + math.pow(point1[1] - point2[1],2) + math.pow(
                point1[2] - point2[2], 2))
        #print 'distance: '+ str(dist)
        return dist

In this program it will return the distance in mm or inches?

MartyG-RealSense, 2022-07-02:
The RealSense SDK's **get_distance()** instruction returns distance in meters.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1depth__frame.html#a5090d69d04ade6dd67175cae9b6bee9c

closed 2022-07-08 by saisatheesh5