realsenseai/librealsense #10643: realsense d435, raspberry  - No device connected

SmirnovL, 2022-07-01:
Hi, i have realsense d435, raspberry 4 and the program, it runs in a loop, receives frames, but sometimes 1 time a day, 1 time in 2 days it freezes with the error "No device connected", restarting the program does not help, only restarting the device, which may be associated with?
Thanks for answer!

MartyG-RealSense, 2022-07-01:
Hi @SmirnovL  If a RealSense 400 Series camera in a long-run session remains within its recommended temperature range then if a problem occurs hours / days later the cause will typically be one that is not related to the camera hardware.   This could include a glitch on the USB port / USB cable, or a problem on the computer hardware or its operating system.

The official recommended maximum operating temperature is 35 degrees C, though the camera should be able to operate normally at up to 42 degrees C in real-world conditions before it may begin to experience problems above that temperature.

It may be helpful to perform a long-run session with the camera in the RealSense Viewer tool instead of your own program and monitor the **ASIC** and **Projector** temperatures to see whether they rise substantially over time.

SmirnovL, 2022-07-07:
@MartyG-RealSense how i can get temperature from python command? the camera is located in a small box with a raspberry and on a raspberry the temperature is 55 degrees

SmirnovL, 2022-07-07:
```
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)

cfg = pipeline.start(config)
depth_sensor = cfg.get_device().first_depth_sensor()

# Get device temperature
tempProj = depth_sensor.get_option(rs.option.projector_temperature)
print("tempProj: " + str(tempProj))
tempAsic = depth_sensor.get_option(rs.option.asic_temperature)
print("tempAsic: " + str(tempAsic))
```

SmirnovL, 2022-07-07:
@MartyG-RealSense hmm 54-58С, its not ok? =)

MartyG-RealSense, 2022-07-07:
54-58 degrees C is not a healthy camera temperature and there is a good chance of problems occurring at that temperature range.  That kind of internal temperature may occur if the natural temperature of the location that the camera is being used in is strongly heating the camera casing, or if there is an electrical problem with the USB port or USB cable that is causing overheating.

If the camera casing is hot to the touch after only a few minutes of activating from a cold start then that would be a good indicator that there may be a USB electrical problem.

closed 2022-07-19 by MartyG-RealSense