realsenseai/librealsense #10645: read infrafed from BAG

L-xn, 2022-07-03:
I would like to ask how to read BAG's infrared video stream in Python and save it as AVI.I have implemented saving depth streams and color streams as AVI.I want to use Tensorflow to train my deep data.

MartyG-RealSense, 2022-07-03:
Hi @L-xn  Another RealSense user at https://github.com/IntelRealSense/librealsense/issues/5823 who was also doing deep learning attempted to use Python to save infrared to .avi.  Like yourself, they were able to save color and depth but not infrared.  They share their script in the discussion.

If you are using the infrared Y8 format for your IR stream, please first try changing its configuration to **Y16** to confirm whether or not the problem with recording infrared to .avi with OpenCV is related to the bit format (8-bit for Y8 and 16-bit for Y16).  For example:

`config.enable_stream(rs.stream.infrared, 640, 480, rs.format.y16, 15)`

I use '15' as the FPS because Y16 supports a smaller range of FPS modes than Y8.

L-xn, 2022-07-03:
When I record in Intelrealsenseviewer, the format I use is Y8. If I use Y16, the program would report an error. The following is my program, there is an error after running:Frame didn't arrive within 5000.
`#####################################################
##               Read bag from file                ##
#####################################################

# First import library
import pyrealsense2 as rs
# Import Numpy for easy array manipulation
import numpy as np
# Import OpenCV for easy image rendering
import cv2
# Import argparse for command-line options
import argparse
# Import os.path for file path manipulation
import os.path

# Create object for parsing command-line options（为解析命令行选项创建对象）
parser = argparse.ArgumentParser(description="Read recorded bag file and display depth stream in jet colormap.\
                                Remember to change the stream fps and format to match the recorded.")
# Add argument which takes path to a bag file as an input（添加参数，它接受一个包文件的路径作为输入）
parser.add_argument("-i", "--input", type=str, help="Path to the bag file")
# Parse the command line arguments to an object（将命令行参数解析为一个对象）
args = parser.parse_args()
# Safety if no parameter have been given（安全，如果没有给出参数）
if not args.input:
    print("No input paramater have been given.")
    print("For help type --help")
    exit()
# Check if the given file have bag extension
if os.path.splitext(args.input)[1] != ".bag":
    print("The given file is not of correct file format.")
    print("Only .bag files are accepted")
    exit()
try:
    # Create pipeline
    pipeline = rs.pipeline()
    # Create a config object
    config = rs.config()
    # Tell config that we will use a recorded device from file to be used by the pipeline through playback.
    #（告诉config，我们将使用一个记录设备从文件到管道使用通过播放。）
    rs.config.enable_device_from_file(config, args.input,repeat_playback=False)
    # Configure the pipeline to stream the depth stream（将管道配置为深度流）
    # Change this parameters according to the recorded bag file resolution（请根据记录的包文件分辨率修改此参数）
    #config.enable_stream(rs.stream.depth, rs.format.z16, 30)

    config.enable_stream(rs.stream.infrared,1280,720,rs.format.y8,30)
    config.enable_stream(rs.stream.infrared,2, 1280, 720, rs.format.y8,30)

    infrared_path1='H:/uettrain/038_1.avi'
    infrared_path2 = 'H:/uettrain/038_2.avi'

    infrared1writer = cv2.VideoWriter(infrared_path1, cv2.VideoWriter_fourcc(*'XVID'), 30, (1280, 720),1)
    infrared2writer = cv2.VideoWriter(infrared_path2, cv2.VideoWriter_fourcc(*'XVID'), 30, (1280, 720),1)

    # Start streaming from file
    profile=pipeline.start(config)

    playback = profile.get_device().as_playback()
    playback.set_real_time(False)
    # Create opencv window to render image in
    #cv2.namedWindow("Depth Stream", cv2.WINDOW_AUTOSIZE)
    cv2.namedWindow("Depth Stream", cv2.WINDOW_FREERATIO)

    # Streaming loop
    while True:
        profile.get_device().as_playback().resume()
        frames = pipeline.wait_for_frames()
        profile.get_device().as_playback().pause()
        frames.keep()

        # Get frameset of depth

        infra1 = frames.get_infrared_frame(1)
        infra2 = frames.get_infrared_frame(2)

        infra1_image=np.asanyarray(infra1.get_data())
        infra2_image = np.asanyarray(infra2.get_data())

        infrared1writer.write(infra1_image)
        infrared2writer.write(infra2_image)

        D_Cimage = np.vstack((infra1_image, infra2_image))

        # Render image in opencv window
        cv2.imshow("Depth Stream", D_Cimage)
        key = cv2.waitKey(1)
        # if pressed escape exit program
        if key == 27:
            cv2.destroyAllWindows()
            break

finally:

    infrared1writer.release()
    infrared2writer.release()
    pipeline.stop()
    pass
`

MartyG-RealSense, 2022-07-03:
I located another RealSense Python script that saves IR to .avi.  You can find it if you scroll down to the last script in the article at the bottom of the page at the link below, which has the title ** Open CV and Numpy integration** in its header section.

https://qiita.com/nataly510/items/c676a28a1119907a560d

Is tthis script helpful for you, please?

[image]

MartyG-RealSense, 2022-07-09:
Hi @L-xn  Do you require further assistance with this case, please?  Thanks!

L-xn, 2022-07-10:
Thank you very much! I will continue to study!

MartyG-RealSense, 2022-07-10:
Thanks very much @L-xn for the update - good luck!

MartyG-RealSense, 2022-07-22:
In the small number of cases where people have tried to save infrared to video with VideoWriter, they ended up with empty or corrupted files.

The reference that is the most potentially useful is the one at the link below.

https://stackoverflow.com/questions/71945367/how-to-properly-use-cv2-videowriter-to-write-depth-video-and-infrared-video-simi

In that discussion they were able to achieve a playable video with **cv2.VideoWriter_fourcc(*'mp4v')**

closed 2022-08-03 by MartyG-RealSense