realsenseai/librealsense #10646: Frame didn't arrive within 5000

L-xn, 2022-07-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Error with language circled in red:Frame didn't arrive within 5000 .Is it because my videos are so short? Only 19 seconds.
[image]

MartyG-RealSense, 2022-07-03:
Hi @L-xn  The **Keep()** instruction stores frames in the computer's memory and so the computer's memory capacity can be quickly consumed as time progresses.  The duration of recorded data that can be stored will depend on the memory capacity.  For example, a computing device such as Raspberry Pi with a memory size such as 4 GB may be able to store 10 seconds worth of frames, whilst a computer with a larger memory capacity such as a desktop / laptop PC could store a longer period such as 30 seconds. 

Does the **Frame didn't arrive within 5000** error occur immediately after the program starts or after about 5 seconds have passed?

L-xn, 2022-07-03:
An error occurred about 6 seconds after executing the program:Frame didn't arrive within 5000

MartyG-RealSense, 2022-07-03:
That suggests to me that it is not a problem with running out of memory and that frames are not arriving when the program starts and so the camera connection time-outs after 5 seconds of not receiving frames.

Do you still experience the error if you remove the second wait_for_frames instruction highlighted in white in the image below?

[image]

L-xn, 2022-07-03:
Thanks!This issue has been resolved!

MartyG-RealSense, 2022-07-03:
That's great to hear, @L-xn - thanks very much for the update!

closed 2022-07-09 by MartyG-RealSense