realsenseai/librealsense #10647: How to grab frames and Save .MP4 or .AVI videos from Real Sense Camera (d435 Series) in python with OpenCV?

Jondila, 2022-07-03:
I am using a real sense camera to grab and save infrared frames for the format mentioned in the subject line above. I have also served some tutorials that helped me initialize the frames however saving the video frames becomes a challenge. I was able to save the video but when I try to play it through the media player, the .avi file indicates that it's corrupted.
Here are the codes used for the camera for the saving problem

```
import pyrealsense2 as rs
import numpy as np
import time
import cv2
import os

os.chdir('D:/kjn')
device = rs.context().devices[0]
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)
profile = pipeline.start(config)
#print(device)
device.get_info(rs.camera_info.advanced_mode)
device.sensors[0].profiles

# streaming loop
try:
    #out = cv2.VideoWriter('./labtest_video.avi',cv2.VideoWriter_fourcc('M','J','P','G'), 30, (640,480))
    out1 = cv2.VideoWriter('./labtest1_video.avi',cv2.VideoWriter_fourcc(*'DIVX'), 30, (640,480))
    while True:
        frames = pipeline.wait_for_frames()
        left_frame = frames.get_infrared_frame(1) # Left IR Camera, it allows 1, 2, no input
        if not left_frame:
            continue
        images = np.asarray(left_frame.get_data())
        out1.write(images)
        cv2.imshow('infrared', images)
        k = cv2.waitKey(1)
        if k==27:
            print('ESC')
            cv2.destroyAllWindows()
            break
finally:
    # Stop streaming
    pipeline.stop()
```
Additionally, how to remove the what points/patterns on the infrared frames when the real sense camera is used in a room with zero illumination?

MartyG-RealSense, 2022-07-03:
Hi @Jondila  Given that issues with recording RealSense infrared to .avi have already been reported in https://github.com/IntelRealSense/librealsense/issues/10645 - could you try running the color and depth .avi saving script at https://github.com/IntelRealSense/librealsense/issues/2731#issuecomment-529938267 to see whether the resulting depth and color .avi file can play in your media player, please?

You can remove the dot pattern on the infrared frame by disabling the projector component of the camera and its **IR emitter** using Python code such as the script at https://github.com/IntelRealSense/librealsense/issues/10065 to set the instruction **rs.option.emitter_enabled** to a value of '0'.

Alternatively, you could keep the projector on but make the pattern invisible on the infrared image by setting **Laser Power** to a value of '1' with the instruction **rs.option.laser_power** (setting it to '0' disables the projector), as demonstrated in a Python script at https://github.com/IntelRealSense/librealsense/issues/8592#issuecomment-800257572

If you require illumination in a zero illumination environment then the camera can make use of light from an external **IR illuminator** lamp. Information about this is at https://github.com/IntelRealSense/librealsense/issues/2000

Jondila, 2022-07-03:
> 

I am able to save the color and depth videos through python OpenCV and I can play them through a media player. I tried to record for 3 hrs and found that the depth and colored videos had more than 3 GBs while the IR video frames had just 5.5KB only.

Jondila, 2022-07-03:
It is possible to remove the dot pattern on the infrared by all the options provided above. But it is not possible for saving an infrared video after following the comments from the first paragraph of the comment above @MartyG-RealSense

MartyG-RealSense, 2022-07-03:
I located another RealSense Python script that saves IR to .avi.  You can find it if you scroll down to the last script in the article at the bottom of the page at the link below, which has the title ** Open CV and Numpy integration** in its header section.

https://qiita.com/nataly510/items/c676a28a1119907a560d

Does this code work for you, please?

[image]

Jondila, 2022-07-04:
This code works properly on both depth and color frames, but when I change them for infrared and depth that is where the problem exists with the IR video frames. The saved file is just 5.5KB or sometimes 14KB and it seems to be corrupted. Let me try to follow the link in your comment and I will let you know of the  progress @MartyG-RealSense

MartyG-RealSense, 2022-07-04:
Thanks very much @Jondila - please do update me with your progress.  Good luck!

Jondila, 2022-07-05:
> Thanks very much @Jondila - please do update me with your progress. Good luck!

Thank you for sharing with me the two links in the comments above, that I have used and I have the following observations.
From the first alternative, I was supposed to save a **.bag** file and I can use the second piece of code to read the **.bag** file frames, however,
a) when I separate the .bag file into depth, infrared, and colored, and then write them to the local disc independently, only the colored one had the 185MB while the rest had 5.5KB (seems to be corrupted)
b) When I separate the actual frames without saving the **.bag**  format, the colored and depth frames were having playable content in the media player while the infrared had 5.5KB and sometimes 0KB (because I have done it repetitively).

My Question. Is there any format suitable for saving (writing) infrared frames because of its pixel value format (y8 or y16)? . I have tried all the format values but I haven't yet solved the existing problem

I have used the codes from the shared links

MartyG-RealSense, 2022-07-05:
On a non-RealSense case at the link below, someone tried expressing their fourcc videowriter instruction as **cv2.VideoWriter.fourcc('Y','1','6',' '))** to obtain Y16.

https://flir.custhelp.com/app/answers/detail/a_id/3387/~/boson-video-and-image-capture-using-opencv-16-bit-y16

MartyG-RealSense, 2022-07-11:
Hi @Jondila  Do you require further assistance with this case, please?  Thanks!

Jondila, 2022-07-14:
> Hi @Jondila Do you require further assistance with this case, please? Thanks!
Hi MartyG, I still need assistance writing a real sense camera's IR (real-time) frames. One time I tried to write with MP4 fourcc format, at least its content had 188KB but it wasn't playable content (corrupted content) on the Media Player.
Thank you for checking up. @MartyG-RealSense

MartyG-RealSense, 2022-07-14:
I researched the subject again but found the same conclusions.  In the small number of cases where people have tried to save infrared to video with VideoWriter, they ended up with empty or corrupted files.

The reference that is the most potentially useful is the one at the link below.

https://stackoverflow.com/questions/71945367/how-to-properly-use-cv2-videowriter-to-write-depth-video-and-infrared-video-simi

In that discussion they were able to achieve a playable video with **cv2.VideoWriter_fourcc(*'mp4v')**

Jondila, 2022-07-17:
Thank you for your sincere help.
I am still trying to solve with multiple options and I might come up with a solution soon.
@MartyG-RealSense

MartyG-RealSense, 2022-07-17:
Thanks very much @Jondila for the update - good luck!

MartyG-RealSense, 2022-08-07:
Case closed due to no further comments received.

closed 2022-08-07 by MartyG-RealSense

DukeMacrocat, 2022-11-30:
Hellom, i am new here please i have a Question about intel RealSense d455. Can i use it to record multi-videos with same frames but using only one intel RealSense d455 camera?  i want to use the videos for 3D and calibrations.
Please help.
Thank you in advance.

MartyG-RealSense, 2022-11-30:
Hi @DukeMacrocat  When you say 'multi-videos', do you want to record different types of stream (depth, RGB, infrared, etc) at the same time with a single camera?   You could do that by recording a **bag** file, which is like a video recording of camera data.  An advantage of a bag file is that a program can read data from it as though it is a live camera.  

A simple way to record multiple streams to a bag file would be to use the **RealSense Viewer** tool to enable the stream types that you want to use and then click the Viewer's **Record** button to start and stop the recording.

DukeMacrocat, 2022-11-30:
@MartyG-RealSense Thank you for your reply.
I mean i want to record 2 or more videos for my 3D project. i am working on this github {https://github.com/lambdaloop/anipose} but for them they used 3 cameras to record there videos. I thought because intel Real Sense d455 camera has more than 1 camera i can use it to record the videos i need. please is it possible to use only one d455 camera.
Thank you.

MartyG-RealSense, 2022-11-30:
As mentioned earlier in this discussion, https://github.com/IntelRealSense/librealsense/issues/2731#issuecomment-529938267 has a Python script for saving depth and RGB from a single camera as separate .avi format videos.

DukeZacharia, 2022-12-01:
@MartyG-RealSense i did try it out using infrared 1 and 2 the camera is well obtained but it returns empty viedio.aiv files. i have tried the above recommended methods but i can get any that works for me.

MartyG-RealSense, 2022-12-01:
@DukeZacharia  Whilst saving to depth and RGB to .avi works, saving infrared to .avi tends to have more problems related to empty or corrupted .avi files, as referenced above at https://github.com/IntelRealSense/librealsense/issues/10647#issuecomment-1174923523 and https://github.com/IntelRealSense/librealsense/issues/10647#issuecomment-1184501322

Some additional scripting references for saving infrared to video are at https://github.com/IntelRealSense/librealsense/issues/10645