realsenseai/librealsense #10648: wrappers/tensorflow/example4 - train.py

L-xn, 2022-07-03:
The error shown in the following figure occurs when I run this code.
`import os, sys, shutil
import keras
import time
import tensorflow as tf
from keras import backend as kb
from keras.models import *
from keras.layers import *
from keras.optimizers import *
from keras.callbacks import ModelCheckpoint
import glob
from PIL import Image
import numpy as np
import cv2
from skimage import img_as_uint

#============================================ T R A I N   M A I N ========================================
root = r"./unet_flow"
images_path = root + r"/images"
models_path = root  + r"/models"
logs_path = root  + r"/logs"
train_images = images_path + r"/train"
train_cropped_images_path = images_path + r"/train_cropped"
paths = [root, images_path, models_path, logs_path, train_images, train_cropped_images_path]

print("Creating folders for training process ..")
for path in paths:
    if not os.path.exists(path):
        print("Creating  ", path)
        os.makedirs(path)

#============================================ T R A I N ===============================================
# other configuration
channels = 2
img_width, img_height = 128, 128

unet_epochs = 1

IMAGE_EXTENSION = '.png'
# Get the file paths
kb.clear_session()
gpus = tf.config.experimental.list_physical_devices('GPU')
print("Num GPUs Available: ", len(tf.config.experimental.list_physical_devices('GPU')))
if gpus:
    try:
        # Currently, memory growth needs to be the same across GPUs
        for gpu in gpus:
            tf.config.experimental.set_memory_growth(gpu, True)
        logical_gpus = tf.config.experimental.list_logical_devices('GPU')
        print(len(gpus), "Physical GPUs,", len(logical_gpus), "Logical GPUs")
    except RuntimeError as e:
        # Memory growth must be set before GPUs have been initialized
        print(e)

print("Creating and compiling Unet network .. ")

# save output to logs
old_stdout = sys.stdout
timestr = time.strftime("%Y%m%d-%H%M%S")
model_name = 'DEPTH_' + timestr + '.model'
name = logs_path + r'/loss_output_' + model_name + '.log'
log_file = open(name, "w")
sys.stdout = log_file
print('Loss function output of model :', model_name, '..')

input_size = (img_width, img_height, channels)
inputs = Input(input_size)
conv1 = Conv2D(64, 3, activation='relu', padding='same', kernel_initializer='he_normal')(inputs)
conv1 = Conv2D(64, 3, activation='relu', padding='same', kernel_initializer='he_normal')(conv1)
pool1 = MaxPooling2D(pool_size=(2, 2))(conv1)
[... 329 lines cut ...]
        row, col = int(row), int(col)
        denoised_image = model.predict(sample)
        row_end = row + cropped_h
        col_end = col + cropped_w
        denoised_row = cropped_h
        denoised_col = cropped_w
        if row + cropped_h >= height:
            row_end = height-1
            denoised_row = abs(row-row_end)
        if col + cropped_w >= width:
            col_end = width-1
            denoised_col = abs(col - col_end)
        # combine tested images
        whole_image[row:row_end, col:col_end]=  denoised_image[:, 0:denoised_row,0:denoised_col, :]
    t2 = time.perf_counter()
    print('test: ', os.path.basename(directory.split('/')[-1]), ': ', t2 - t1, 'seconds')
    denoised_name = os.path.basename(directory.split('/')[-1])
    outfile = denoised_dir + '/' + denoised_name.split('-')[0] + '' + '_denoised-' + denoised_name.split('-')[1] + IMAGE_EXTENSION
    whole_image = img_as_uint(whole_image)
    cv2.imwrite(outfile, whole_image[:,:,0])
sys.stdout = old_stdout
log_file.close()
print("Testing process is done successfully !")
[image]
`

L-xn, 2022-07-03:
[image]

MartyG-RealSense, 2022-07-03:
Hi @L-xn  The line of **example 4 - train.py** that the trace log indicates that the prob;em is located at this the one lin the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/tensorflow/example4%20-%20train.py#L221

May I first confirm with you please whether you are using **Python 3.6**, which is recommended by the **Installtion** section of the TensorFlow wrapper due to its compatibility with the components used in the wrapper?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow#installation

Have you also downloaded both parts 1 and 2 of the dataset for Example 4, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow#training-dataset

L-xn, 2022-07-04:
My configuration is:python3.9,conda11.6,tensorflow-gpu2.9.1.I just put part1 in the program's images directory.The types of pictures in Parts 1 and 2 are the same.Below is the log of my running program.
[image]

L-xn, 2022-07-04:
[image]

MartyG-RealSense, 2022-07-04:
I have studied train.py and the error logs carefully.  The **cannot reshape array of size 1000** problem seems to be occurring around the area of the train.py script highlighted in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/tensorflow/example4%20-%20train.py#L167-L169

The number of iterations is being reported by the program output as being '0'.  The value for iterations is calculated from the number of cropped images.  If it is reporting 0 iterations then I wonder whether it is having difficulty finding the folder path where the trapped images are located, or if it can find the path then no cropped versions of the images have been generated.

MartyG-RealSense, 2022-07-09:
Hi @L-xn  Do you require further assistance with this case, please?  Thanks!

L-xn, 2022-07-10:
Thank you very much !

MartyG-RealSense, 2022-07-10:
If you are satisfied with the outcome of the case then please feel free to close it using the **Close Issue** button underneath the comment writing box.  If you have further questions though then I will be happy to help.

closed 2022-07-16 by MartyG-RealSense