realsenseai/librealsense #10650: Connect to D455e

Nileshhampiholi, 2022-07-04:
| Required Info                         |    Connect to device                              |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D455e }                                                | 
| Operating System & Version |   {Win (10)                                               | 
| Platform                                 | PC                                                           |
| Language                               |  python                                                  | 

### Issue Description

I have a industrial sensor D455e. I want to stream colour and depth frames and generate point clouds. But I am not able to connect to the device. I am using python  I could not find any documentation as well. 

Thank you

MartyG-RealSense, 2022-07-04:
Hi @Nileshhampiholi  Support for the 'e' series of RealSense industrial cameras is provided by the **FRAMOS** company that produces them.  The dedicated SDK software, firmware drivers and documentation provided by FRAMOS at the support page link below should be used.

https://www.framos.com/en/industrial-depth-cameras#downloads

closed 2022-07-11 by MartyG-RealSense

MartyG-RealSense, 2022-07-11:
Hi @Nileshhampiholi  I hope that you were able to find the information for your FRAMOS camera that you were seeking.

Nilesh-Hampiholi, 2022-07-12:
> Caution: This e-mail originated from outside of Philips, be careful for phishing.

Hey Marty,

I found the way I need to build the realsense from source. But I have having problems with that as well.

***@***.***
Can you suggest any solution.
I have already tried reinstall MSVC and SDK

Regards
Nilesh

The information contained in this message may be confidential and legally protected under applicable law. The message is intended solely for the addressee(s). If you are not the intended recipient, you are hereby notified that any use, forwarding, dissemination, or reproduction of this message is strictly prohibited and may be unlawful. If you are not the intended recipient, please contact the sender by return e-mail and destroy all copies of the original message.

MartyG-RealSense, 2022-07-12:
I recommend visiting the FRAMOS website at the link below and moving your mouse cursor to the bottom of the page until the **Let's Talk** button appears, which opens a web-chat box when clicked on so that you can talk to a FRAMOS support representative.  

https://www.framos.com/en/