realsenseai/librealsense #10651: Getting resolution and framerate of recorded D400 bag files

JeffR1992, 2022-07-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |   Win 11 / Linux (Ubuntu 18)  | 
| Platform                                 | PC |
| Language                            |  Python                          | 

### Issue Description
I want to be able to play back recorded bag files for analysis, but do not know at what resolution or framerate each bag file was recorded at (and each bag file could have been recorded at different resolutions and framerates). Here is an example where I'm enabling the left infrared stream and the depth stream, but I've hard coded the resolution to 1280x720, and the framerate to 30.

```
config = rs.config()
config.enable_device_from_file(file_path, repeat_playback=False)
config.enable_stream(rs.stream.infrared, 1, 1280, 720, rs.format.y8, 30)
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
```

Is there a way where I can actually find out that the infrared and depth streams were recorded at a resolution of 1280x720 and at a framerate of 30 fps, instead of hard coding it?

MartyG-RealSense, 2022-07-05:
Hi @JeffR1992  You can quickly find the resolution and FPS of streams in a specific bag file by dragging and dropping the bag file into the center panel of the RealSense Viewer tool to play it back and then expanding open the Stereo Module and RGB sections of the Viewer's options side-panel.  Shown below are the resolution and FPS of a few of the test bags from the SDK's 'Sample Data' page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

**outdoors.bag (D415)**

[image]

**stairs.bag (D435)**

[image]

**structured.bag (SR300)**

[image]

MartyG-RealSense, 2022-07-18:
Case closed due to no further comments received.

closed 2022-07-18 by MartyG-RealSense

SammaniSandumini, 2024-02-12:
Hi @MartyG-RealSense . When I open the Stereo Module and RGB sections of the Viewer's options side-panel as you say it shows 640x480 ,  But in frame metadata section  of stereo module depth stream it shows, display size as 320x240. but in color stream it shows correctly as 640x480.why is that. what is the difference between hardware size and display size?(I have recorded the bag files in resolution 640x480 and frame rate=30)

[image]

MartyG-RealSense, 2024-02-12:
Hi @SammaniSandumini  When the **Decimation Filter** post-processing filter is enabled (which it is by default in the Viewer), the currently selected depth resolution is halved by a factor of two.  You can disable the filter in the 'Post Processing' section of the Viewer's options side-panel by left-clicking on the blue icon beside the Decimation Filter to disable it.  The overlay should then show the resolution as the one that was selected (640x480 in this case).

SammaniSandumini, 2024-02-12:
Thank you very much for your quick reply. I have another question, although I targeted 60fps, recordings only reach around 30fps. Why is that?

SammaniSandumini, 2024-02-12:
Auto Exposure Priority option in the RGB section is disabled. Here Hardware FPS = 59.53, Viewer FPS varies between 31-35, Actual Fps = 60 .

MartyG-RealSense, 2024-02-12:
If you click the 'i' icon at the top of each stream panel then it will overlay the real-time FPS for that stream at the top of its panel.  This is the value that you should rely on.

SammaniSandumini, 2024-02-12:
Got it, thanks for pointing that out!