realsenseai/librealsense #10652: Couldn't get the edges of Depth Image when using MVTec Halcon

amar-mustaqim, 2022-07-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  (n.a.)                                         | 
| Platform                                 | PC |
| Language                            |  {Halcon}                          | 
| Segment			|  {Others}                          | 

### Issue Description
Hi there, we have a Realsense Depth Camera D455 that we plan to use for checking the dimension of the box. We set the camera at the height of 2meter and plan to capture the depth image of the box, and from the depth image we are going to do some image processing later. Currently, we are using MVTec Halcon(Realsense interface) for our development and here is our result. The red circle is the box that we target to capture. 

[image]

These are the parameter that we use in our implementation:
set_framegrabber_param(AcqHandle, 'SourceSelector', 'Depth')
set_framegrabber_param(AcqHandle, 'ComponentSelector', 'Intensity')
set_framegrabber_param(AcqHandle, 'ComponentEnable', 100)
set_framegrabber_param(AcqHandle, 'DepthResolution', '848x480')
set_framegrabber_param(AcqHandle_3D, 'LaserPower', 360.0)

My question is, how can we improve the result for the box? Is there a way we can get the edge of the box or the exact square shape of it? Any parameter in MVTec Halcon, configuration on the camera or suggestions can be made, please suggest.

Thank you in advance for your help.

MartyG-RealSense, 2022-07-05:
Hi @amar-mustaqim  The color shades of the cardboard box and the floor are similar, so the box does not stand out strongly on the depth image.

Are you able to set a RealSense **Visual Preset** camera configuration from within the MVTec Halcon?  If so then setting the **Medium Density** preset may help to sharpen the edges of a 'blobby' cardboard box on the depth image whilst also providing accuracy, as demonstrated in a discussion about overhead depth-sensing of cardboard boxes at https://github.com/IntelRealSense/librealsense/issues/7021

amar-mustaqim, 2022-07-06:
Hi @MartyG-RealSense , I have tried to test with different color background which are white and dark blue color, and I tried the Medium Density preset setting as suggested. However the result still same as before, no edges appeared for the box at the distance of 2meter from the camera. 

Result of dark blue color background:
[image]

Result of white color background:
[image]

Result with Medium Density preset:
[image]

MartyG-RealSense, 2022-07-06:
Are you able to use auto-exposure instead of a fixed manual exposure value of 33000?  If you can use auto-exposure then please try setting **ExposureAuto** to On, as the current Off status of that setting suggests that auto-exposure is disabled.

amar-mustaqim, 2022-07-08:
The **ExposureAuto** is already set to On by default in MVTec Halcon during the image acquisition, so even though I'm using auto-exposure the edge of the image still not captured. That's why I try changing the manual exposure value, which automatically will disable the **ExposureAuto** but, the result still same.

MartyG-RealSense, 2022-07-08:
The depth images look very dark even though the RGB images show that there is good illumination.  Could you try setting **EmitterAlwaysOn** to On (it is Off in the image of the settings) to make sure that the camera's dot pattern projection is active.

MartyG-RealSense, 2022-07-14:
Hi @amar-mustaqim  Do you require further assistance with this case, please?  Thanks!

amar-mustaqim, 2022-07-14:
Hi @MartyG-RealSense , yes I'm still need your help for this case but not immediately, maybe at the end of this month I'll come back for this case. As for now, we just lower down our camera setup until it able to get the edge of the box because we are focusing on the algorithm to measure the box dimension first.

MartyG-RealSense, 2022-07-14:
Okay, that's no problem at all to come back to the case at the end of the month.  Thanks very much for the update!

MartyG-RealSense, 2022-08-24:
Hi @amar-mustaqim  Do you have an update about this case that you can provide, please?  Thanks!

amar-mustaqim, 2022-08-25:
Hi @MartyG-RealSense I can say that this issue already solve and I will close it. Thank you so much for your assist. Cheers!

closed 2022-08-25 by amar-mustaqim

MartyG-RealSense, 2022-08-25:
That's great news that you achieved a solution, @amar-mustaqim - thanks very much for the update  :)

vpenev77, 2024-06-04:
Hi, I'm trying to grab a 2d rgb image using realsense and halcon but it doesn't work. Can you help me