realsenseai/librealsense #10654: D455/python

L-xn, 2022-07-05:
Sorry to bother you!Now I have installed ROS, and I want to filter the depth information in BAG through ROS, so as to save it as a filtered BAG.How do I do that in Python?

MartyG-RealSense, 2022-07-05:
Hi @L-xn  What kind of filtering do you wish to perform, please?  The RealSense ROS wrapper has in-built support for some types of post-processing filtering that does not require use of Python code.

[image]

Also, are you recording a **rosbag** file within ROS with the **rosbag record** tool or aiming to use Python code to record ROS topics to an SDK-bag using the librealsense SDK's bag recording function?

L-xn, 2022-07-06:
I had previously recorded BAG using the realsenseviewer.However, the filtering operation inside realsenseviewer does not change BAG.So can I change BAG recorded by RealSenseViewer through ROS?

MartyG-RealSense, 2022-07-06:
The RealSense SDK and tools based on it such as the RealSense Viewer do not save post-processing filtering into SDK-bag files.  Instead, the unfiltered stream data is loaded from the bag and then post-processing can be applied to that loaded-in data in real time.

Whilst the RealSense ROS wrapper can load bag files to use as a data source instead of a live camera and can apply post-processing filtering in real-time, there are differences between the SDK-bag format and the format of a rosbag file recorded in ROS, as advised at https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

A ROS workaround for playback of bag files recorded in the Viewer is suggested at https://github.com/IntelRealSense/librealsense/issues/4260#issuecomment-511878023

The ideal approach would be to record a new bag, in the RealSense ROS wrapper with **rosbag record** this time instead of the Viewer, and then load the ROS rosbag and apply the wrapper's post-processing filters to the bag data.

closed 2022-07-20 by MartyG-RealSense