realsenseai/librealsense #10658: Nothing is streaming as enable "Motion Module" on Jetson Xavier NX with D435i

HEIDIES, 2022-07-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Name                       | D435i  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-177-generic | 
| Platform                                 | NVIDIA Jetson Xavier NX  |
| SDK Version                            |  2.50.0 | 

### Issue Description
I'm testing D435i camera on Jetson Xavier NX using Realsense Viewer. It works property when just enable "Stereo Module" and "RGB Camera". But nothing is streaming after enabling "Motion Module".

MartyG-RealSense, 2022-07-07:
Hi @HEIDIES  It is a known issue that when Depth, RGB and the IMU (Motion Module) are enabled simultaneously then one of the streams (usually RGB) can disappear and be replaced with a **No Frames Received** message.  It does not make a difference which order the three stream types are enabled in, as the enabling of the Motion Module is always the trigger for the issue.

In the RealSense Viewer tool there is no fix other than to use only two streams (IMU and depth, or IMU and RGB).  

The RealSense SDK C++ example program **rs-capture** is not affected and can display the depth, RGB and IMU streams simultaneously, though the data is not interactive in that program.

When creating your own script, a workaround fix on Python is to define two separate pipelines and put the IMU on one pipeline on its own and depth + RGB on the other pipeline.  The best available example of this method is at https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238 

For C++, **callbacks** can be used to address the problem.

HEIDIES, 2022-07-11:
Hi @MartyG-RealSense , thanks for your advice.
And I want to know more about why this kind of issue happens. It works fine on x86 system, so I suspect it's a driver problem on Arm system, is that so?

MartyG-RealSense, 2022-07-11:
It occurs on x86 systems too but is not experienced by all RealSense users, suggesting that it may depend upon the particular computer being used.  There is not a known reason for the issue.

HEIDIES, 2022-07-12:
Hi @MartyG-RealSense , I have tested again following your advice. This time, I only enable the "Motion module" on the realsense-viewer, but it still shows "Nothing in streaming".

HEIDIES, 2022-07-13:
I really did, and it's a really issue for me. 
[image]

MartyG-RealSense, 2022-07-13:
Usually if a Motion Module Failure pop-up message occurs in the Viewer then the IMU streams work normally if you then try enabling the Motion Module again after the pop-up has been closed.

HEIDIES, 2022-07-14:
Actually, I have tried many combinations and had the results as follows:
0.enable "Stereo Module" and "RGB Camera" on realsense-viewer, success.
1.enable "Stereo Module", "RGB Camera" and "Motion Module" on realsense-viewer, fail.
2.enable "Stereo Module" and "Motion Module" on realsense-viewer, fail.
3.enable "RGB Camera" and "Motion Module" on realsense-viewer, fail.
4.only enable "Motion Module" on realsense-viewer, fail.
5.enable "Motion Module", disable it and re-enable on realsense-viewer, fail.
6.run "rs-capture" example, fail.
7.try start two different pipelines in my own code, one for "Stereo Module" and "RGB Camera", one for "Motion Module", failed with "No Device Connected" message.

MartyG-RealSense, 2022-07-14:
A couple of Xavier NX users with the Motion Module Failure error have reported that flashing their Jetson resolved the problem.

https://support.intelrealsense.com/hc/en-us/community/posts/4744980703379/comments/4775644497427

HEIDIES, 2022-07-15:
Hi @MartyG-RealSense , I have re-installed the whole library following the instruction, and every works well now.

MartyG-RealSense, 2022-07-15:
That's great news, @HEIDIES - thanks very much for the update  :)

closed 2022-07-17 by HEIDIES

tugbakara, 2022-07-30:
> Hi @MartyG-RealSense , I have re-installed the whole library following the instruction, and every works well now.

Hi Heidies, I got stuck with the same problem with 2.50, could you write whole instraction here. I repeated again again and again. I am in the point that I will give up. I cannot take IMU and Gyro and also cannot calibrate with python script.

MartyG-RealSense, 2022-07-30:
Hi @tugbakara  Could you create a new case please by visiting the front page of this support forum at the link below and clicking the green **New Issue** button please and I will be pleased to help you.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/