realsenseai/librealsense #10659: null pointer passed for argument "frame_ref"

Amayess, 2022-07-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 

| Operating System & Version |   Win 10 | 
| Platform                                 | PC | 
| SDK Version                            | 2.50.0  | 
| Language                            |  C++                        | 
| Segment			|  Robot  | 

### Issue Description

Hi, I am trying to generate a point cloud with by the following code but I got an error saying :  null pointer passed for argument "frame_ref"

Could you please help me? 

[image]

Amayess, 2022-07-08:
Hi @MartyG-RealSense, thank you for the your reactivity ! 

If I do **depth = AverageFrame[i].get_depth_frame();** the three frames i get will not be stored in my Tab AverageFrames[i], non ? 

My objectif here is to recover three frames then store them in my T**ab AverageFrames[i]** so I can only work on them.

MartyG-RealSense, 2022-07-08:
It does not look as though the commented line is vital to the script.  How I read the logic is:

1.  If the number of times that **x = 0; x < W; x++)** has looped is less than the value of W (e.g W = 640, from 640x480) then move onto the next line.

2.  If the number of times that **y = 0; y < H; y++)** has looped is less than the value of H (e.g H = 480, from 640x480) then move on to the next line.

3.  Set the value of Average Depth to 0.

4.  If the number of times that k has looped is less than the value of nb_frames then go to the next line.

**********

Basically, nesting multiple 'for' conditions within each other makes it more likely to go wrong.  It does not look as though the program would be significantly impacted if the **for y = 0; y < H; y++)** line is removed or commented out.

Amayess, 2022-07-11:
Issue Fixed ! thanks @MartyG-RealSense

closed 2022-07-11 by Amayess

MartyG-RealSense, 2022-07-11:
Great, thanks for the update @Amayess    :)