realsenseai/librealsense #10660: Can not get camera SN number when over 5 realsense camera plug in via usb

Wilson-shuo, 2022-07-07:
[
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        |  D455                | 
| Firmware Version                    |  2.5.0                 | 
| Operating System & Version  |  ubuntu18.04    | 
| Kernel Version (Linux Only)     | jetpack 4.5.1     | 
| Platform                                  | NVIDIA Jetson  |
| SDK Version                             | 2.5.0                 | 
| Language                                |  Cpp/Qtcreater | 
| Segment		         	 | Slam                  | 

### Issue Description
Hello everybody!

I pluged in six Realsense D455 cameras on my jetson Nx, but only 5 cameras could be found by the realsense-viewer.

And i try to use cpp and qt to build an app to refresh the camera list, also get only 5 camera.

I was pretty sure that all camera works good, if i unplug one of these 5 camera, the number 6 camera could be found.

so how many  realsense camera could be linked to one cumputer?

Wilson-shuo, 2022-07-07:
[image]
)

Here is my usb list, we can see six realsense device.

Wilson-shuo, 2022-07-07:
[image]

As well as my QT app,I could find five camera,but not six.

MartyG-RealSense, 2022-07-08:
Hi @Wilson-shuo  There was a past Jetson case with a similar situation at https://github.com/IntelRealSense/librealsense/issues/5828 where 6 cameras were attached but only 5 were detected.  In that case, a RealSense user posted at https://github.com/IntelRealSense/librealsense/issues/5828#issuecomment-625671835 a C++ workaround they developed that worked for them in their particular situation.

In regard to how many cameras can be attached to the same computer and operate simultaneously, 6 is typically the maximum.  As that amount of cameras would usually be attached to a USB hub, the average hub will have a data bandwidth limit of 5000 Mbps / 5 Gbps that is shared between all the devices attached to the hub.  So the more cameras that are attached, the greater the chance of the hub's maximum bandwidth limit being reached., especially if multiple simultaneous stream types per camera and / or high resolution & FPS speeds are used.

Intel's multiple camera white-paper guide at the link below provides estimates for bandwidth consumption for up to 6 cameras for various stream configurations.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#2-multi-camera-considerations

It is possible to spread multiple cameras across more than one hub.  The USB standard permits five hubs to be 'daisy chained' together in addition to the 'root' hub.

https://acroname.com/blog/how-many-usb-devices-can-i-connect

MartyG-RealSense, 2022-07-14:
Hi @Wilson-shuo  Do you require further assistance with this case, please?  Thanks!

Wilson-shuo, 2022-07-18:
Thanks for ur reply !

We solve the problem simplely by **reboot the jetson nx**. After reboot,(all camera keep pluged in), the realsense SDK could find six cameras. that is magic!!!

By the  way, we open camera one by one, Only one camera is opened at the same time ,So we do not worry about reach usb max speed.