realsenseai/librealsense #10662: Using librealsense with an RTSP stream from a Kinova Arm

darius-chan-ml, 2022-07-08:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D410 | 
| Firmware Version                   | 2.50.0 | 
| Operating System & Version | Ubuntu 20.04.4
| Kernel Version (Linux Only)    |  5.11.0-27-generic | 
| Platform                                 | PC|
| SDK Version                            | 2.0| 
| Language                            |  opencv + python  | 
| Segment			| Robot | 

Issue Description

Hi,
I'm looking to utilize the realsense library with a realsense camera situated on a Kinova Robotic arm. I've currently connected to the camera using the RTSP stream for both color and depth through GStreamer. I've also been able to save and process images with opencv and GStreamer. However, I'd like to utilize the functionality in pyrealsense2 with the images captured such as align.to and get_distance on depth images. I'm currently running into trouble with working on these images since I am not creating a rs.pipeline to pull frames with. What is the best way to go about utilizing the pyrealsense2 library in this situation?

Is there a method of connecting the realsense pipeline to the gstreamer pipeline? Can I instantiate a frame class with a single image? I've browsed through the documentation, tutorials and issues and haven't found a suitable answer.

Any help is appreciated!

darius-chan-ml, 2022-07-08:
It might be worth noting that the images being streamed from the realsense camera are not through rs-server so I can't connect to the camera with the realsense-viewer unfortunately. The kortex api provided by Kinova offered no functionality to work with the stream directly besides through ROS or the RTSP stream and I won't be able to use ROS in my situation. 

Advice or links would be appreciated!

MartyG-RealSense, 2022-07-08:
Hi @darius-chan-ml  A script that attempts to link pyrealsense2 with GStreamer can be found at the link below.

https://forums.developer.nvidia.com/t/use-intel-realsense-t265-with-gstremer-jatson/165638

If the primary reason for wanting to use pyrealsense2 is to perform depth-color alignment, a RealSense plugin for GStreamer offers alignment as a feature.

https://github.com/WKDSMRT/realsense-gstreamer

darius-chan-ml, 2022-07-08:
Hi Marty,
Thanks for getting back to me! Unfortunately, the first link you provided was a dead end since they closed that issue with no advice. On the realsense plugin for gstreamer, it also presumes the camera is connected and can access the footage via a wired connection.

I think I can move forward with my problem if there is a method of incorporating saved images back into the realsense pipeline. Is that possible with any method in the library?

Thanks!

darius-chan-ml, 2022-07-08:
Alternatively, is it possible to code an RTSP stream as a realsense pipeline?

MartyG-RealSense, 2022-07-08:
If you can save the images as cv::Mat then you may be able to convert them to RealSense **rs2::frame** format using the advice provided by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2634

MartyG-RealSense, 2022-07-14:
Hi @darius-chan-ml  Do you require further assistance with this case, please?  Thanks!

darius-chan-ml, 2022-07-14:
Hi Marty,
I continued looking into the resources provided but there was no simple solution to implementing this challenge. I delved into other forums where the depth image was saved as a .raw file and then streamed in conjunction with a color stream using rs.software_devices to align the frames but I am having difficulty saving a depth image in the right format. If you could direct me to a source that utilizes python to save depth images in a format I can stream, that would be greatly appreciated!

darius-chan-ml, 2022-07-14:
For example: https://github.com/IntelRealSense/librealsense/issues/2231#issuecomment-412804368

MartyG-RealSense, 2022-07-14:
Some Python users have streamed RealSense over the web with a **Flask server**, like the project at the link below.

https://github.com/soarwing52/Remote-Realsense

darius-chan-ml, 2022-07-20:
Hi Marty!
We've currently dropped the issue due to the difficulty of resolving it but will retackle this issue in a few months. Thanks for helping me so far! I hope it'll be alright to reopen this issue down the line.

MartyG-RealSense, 2022-07-20:
Yes, it is fine to close and re-open at a later date.  Good luck!

MartyG-RealSense, 2022-07-26:
Case closed due to plan to re-open it at a future date.

closed 2022-07-26 by MartyG-RealSense