realsenseai/librealsense #10664: undefined reference to "__atomic_load_8"

Marial128, 2022-07-08:
So first sorry for my english (I'm french)
I'm trying to install librealsense on a raspberry PI 3 but when I'm buiding it's show this error :
~/Work/librealsense/build $ make -j1
[ 55%] Built target fw
[ 55%] Built target realsense-file
[ 55%] Built target realsense2
[ 55%] Built target libcurl
[ 55%] Built target glfw
[ 55%] Built target realsense2-gl
[ 55%] Linking CXX executable rs-hello-realsense
/usr/bin/ld : ../../librealsense2.so.2.50.0 : undefined reference to "__atomic_load_8".
/usr/bin/ld : ../../librealsense2.so.2.50.0 : undefined reference to "__atomic_store_8
/usr/bin/ld : ../../librealsense2.so.2.50.0 : undefined reference to "__atomic_fetch_add_8
collect2: error: ld returned 1 exit status
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:110 : examples/hello-realsense/rs-hello-realsense] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:1240 : examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Erreur 2
make: *** [Makefile:149 : all] Erreur 2

If anyone has an idea of what to do ?

MartyG-RealSense, 2022-07-08:
Hi @Marial128  Don't worry, your English is very good  :)

A RealSense user of Raspberry Pi shared their solution for this error at https://github.com/IntelRealSense/librealsense/issues/9962#issuecomment-998392844 that also worked for other Pi users.

Marial128, 2022-07-08:
Thanks for the answer but after doing what he did I still have the same problem : 
[ 55%] Built target libcurl
[ 55%] Built target fw
[ 55%] Built target glfw
[ 55%] Built target rs-embed
[ 55%] Built target realsense-file
[ 55%] Built target realsense2
Scanning dependencies of target rs-capture
[ 55%] Linking CXX executable rs-software-device
[ 55%] Linking CXX executable rs-hello-realsense
[ 55%] Building CXX object examples/capture/CMakeFiles/rs-capture.dir/rs-capture.cpp.o
[ 55%] Built target realsense2-gl
Scanning dependencies of target rs-callback
[ 55%] Building CXX object examples/callback/CMakeFiles/rs-callback.dir/rs-callback.cpp.o
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:110: examples/hello-realsense/rs-hello-realsense] Error 1
make[1]: *** [CMakeFiles/Makefile2:1240: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_fetch_add_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_load_8'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/software-device/CMakeFiles/rs-software-device.dir/build.make:110: examples/software-device/rs-software-device] Error 1
make[1]: *** [CMakeFiles/Makefile2:1268: examples/software-device/CMakeFiles/rs-software-device.dir/all] Error 2
[ 55%] Linking CXX executable rs-callback
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_load_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/callback/CMakeFiles/rs-callback.dir/build.make:110: examples/callback/rs-callback] Error 1
make[1]: *** [CMakeFiles/Makefile2:1324: examples/callback/CMakeFiles/rs-callback.dir/all] Error 2
[ 55%] Linking CXX executable rs-capture
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_fetch_add_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_store_8'
/usr/bin/ld: ../../librealsense2.so.2.50.0: undefined reference to `__atomic_load_8'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/capture/CMakeFiles/rs-capture.dir/build.make:110: examples/capture/rs-capture] Error 1
make[1]: *** [CMakeFiles/Makefile2:1296: examples/capture/CMakeFiles/rs-capture.dir/all] Error 2
make: *** [Makefile:149: all] Error 2

MartyG-RealSense, 2022-07-08:
Can you provide a link to the installation instructions that you used, please?

Marial128, 2022-07-13:
I used the same that devenjarvis posted :

sudo apt install clang llvm
cd ~/librealsense
mkdir build
cd build
export CC=/usr/bin/clang
export CXX=/usr/bin/clang++
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true -DOTHER_LIBS="-latomic" 
make -j4
sudo make install

MartyG-RealSense, 2022-07-13:
The **_atomic_load** error on Pi is a difficult one to debug, unfortunately.

An SDK installation method that has worked well for Pi users in the past is the simple **libuvc backend** one in the link below that builds the SDK with an installation script.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

If the libuvc backend method still has problems with the atomic error, https://github.com/IntelRealSense/librealsense/issues/4950 has advice about modifying the libuvc script to deal with it.

MartyG-RealSense, 2022-07-18:
Is the python-opengl package able to be installed if you input the commands below in the listed order:  

**sudo apt-get update
sudo apt-get -y install python3-opengl**

MartyG-RealSense, 2022-07-19:
I would suspect that retrieving the T265 video via **gstreamer** would be an appropriate solution, like a T265 user in the link below.

https://forums.developer.nvidia.com/t/use-intel-realsense-t265-with-gstremer-jatson/165638

 If you require further assistance for that problem, please create a new issue on this forum.  Thanks!

MartyG-RealSense, 2022-07-25:
Case closed due to separate issue created for T265 issue and no further comments received on this issue.

closed 2022-07-25 by MartyG-RealSense