realsenseai/librealsense #10665: Unable to load IMU data from d435i

KaustubhKhulbe, 2022-07-09:
I am using Intel Realsense D435i Camera, on Mac M1, and trying to use the rs-motion package. Firstly, I ran the rs-hellorealsense package and got it working, meaning the sdk was hooked up. However, the rs-motion gives a ton of open GL errors, specifically it cannot find a bunch of functions. Upon googling, it turned out that it wasn't supported on Mac, so I decided to try and just fetch the imu data.

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2019 Intel Corporation. All Rights Reserved.
#include "../../examples/example.hpp" // Include short list of convenience functions for rendering
#include <cstring>
#include <librealsense2/rs.hpp>
#include <mutex>

rs2::pipeline pipe;

auto cfg = rs2::config();
cfg.enable_stream(RS2_STREAM_GYRO);
cfg.enable_stream(RS2_STREAM_ACCEL);

pipe.start(cfg);

while (true) // Application still alive?
{
  rs2::frameset frameset = pipe.wait_for_frames();

  // Find and retrieve IMU data
  if (rs2::motion_frame accel_frame =
          frameset.first_or_default(RS2_STREAM_ACCEL)) {
    rs2_vector accel_sample = accel_frame.get_motion_data();
    // std::cout << "Accel:" << accel_sample.x << ", " << accel_sample.y << ", "
    // << accel_sample.z << std::endl;
    //...
  }

  if (rs2::motion_frame gyro_frame =
          frameset.first_or_default(RS2_STREAM_GYRO)) {
    rs2_vector gyro_sample = gyro_frame.get_motion_data();
    // std::cout << "Gyro:" << gyro_sample.x << ", " << gyro_sample.y << ", " <<
    // gyro_sample.z << std::endl;
    //...
  }
}
```

This is my code. The error I get is: 
```
/Users/kaustubhkhulbe/Downloads/librealsense-2.50.0/examples/motion/rs-motion.cpp:11:1: error: unknown type name 'cfg'
cfg.enable_stream(RS2_STREAM_GYRO);
^
/Users/kaustubhkhulbe/Downloads/librealsense-2.50.0/examples/motion/rs-motion.cpp:11:4: error: cannot use dot operator on a type
cfg.enable_stream(RS2_STREAM_GYRO);
   ^
/Users/kaustubhkhulbe/Downloads/librealsense-2.50.0/examples/motion/rs-motion.cpp:12:1: error: unknown type name 'cfg'
cfg.enable_stream(RS2_STREAM_ACCEL);
^
```

I'm unsure to what this means, because it seems as though it's not even recognize cfg as a variable, but rather a type?? Any help is much appreciated!

MartyG-RealSense, 2022-07-09:
Hi @KaustubhKhulbe  Please try removing the line **auto cfg = rs2::config();** and instead using the code below for the rs2::config definition and stream configurations.

**rs2::config cfg;
cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F);**

KaustubhKhulbe, 2022-07-09:
Hey @MartyG-RealSense ,
Thanks for the quick reply! That seemed to have fixed that problem. But gl strikes again!

```
Undefined symbols for architecture arm64:
  "_glBegin", referenced from:
      draw_pointcloud(float, float, glfw_state&, rs2::points&) in rs-motion.cpp.o
      draw_pointcloud_wrt_world(float, float, glfw_state&, rs2::points&, rs2_pose&, float*, std::__1::vector<rs2_vector, std::__1::allocator<rs2_vector> >&) in rs-motion.cpp.o
  "_glBindTexture", referenced from:
      draw_pointcloud(float, float, glfw_state&, rs2::points&) in rs-motion.cpp.o
      draw_pointcloud_wrt_world(float, float, glfw_state&, rs2::points&, rs2_pose&, float*, std::__1::vector<rs2_vector, std::__1:
```

Just a ton of these errors. Not really sure how they come, as I'm never using them, but my guess is that they're used in one of my dependancies. Any suggestions?

MartyG-RealSense, 2022-07-09:
A MacOS user who experienced these GL errors suggests a solution at https://github.com/IntelRealSense/librealsense/issues/2685#issuecomment-436498519 for linking to GLFW.

KaustubhKhulbe, 2022-07-09:
Yeah, I've tried to do that. But `g++ -I /usr/local/include/ -I ../ -std=c++11 rs-motion.cpp -o rs-motion -lglfw -lrealsense2     ` just got this error where examples.h doesn't include <librealsense2...>, but my rs-motion does. And *I think* I know why! Basically, I have a folder examples, and underneath it is the folder I'm working on. Since I used brew, I need to do `find_package(realsense2 REQUIRED)` and `target_link_libraries(rs-motion ${realsense2_LIBRARY})`. BUT this only exposes realsense2 in the rs-motion folder, NOT example. I have to modify example's CMakeLists to do so. Thing is... I don't know how.

There isn't really an executable I add it to or anything like that. Could you help me out? The example's CMakeLists is the exact same as librealsense/examples/CMakeLists in the GitHub.

MartyG-RealSense, 2022-07-09:
The set of RealSense SDK examples are built in a different way to how a user-created script is compiled.  On PC at least, they are built as a bundle of examples instead of individually.  When one of the examples is modified, the instruction to build the full bundle of examples is run again, but it only builds the programs that have been modified.

An example of this is the SDK's **rs-hello-realsense** example and an example for building your own independent librealsense program.  The two examples essentially are the same C++ script but the version of the program that is designed to be built independently has a more complex CMakeLists.txt file.

**rs-hello-realsense - CMakeLists.txt**
https://github.com/IntelRealSense/librealsense/blob/master/examples/hello-realsense/CMakeLists.txt

**Getting Started - more complex CMakeLists.txt**
https://github.com/zivsha/librealsense/blob/getting_started_example/examples/getting-started/CMakeLists.txt

KaustubhKhulbe, 2022-07-09:
So by using the more complex CMakeLists, I'll be able to use all the gl stuff and examples stuff?

MartyG-RealSense, 2022-07-09:
I would nomally say Yes, but my knowledge of building on MacOS is limited.

An official librealsense installation instruction page for MacOS indicates that if you are using brew then you can use the instruction **brew install homebrew/core/glfw3**

https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk

KaustubhKhulbe, 2022-07-09:
Thanks! I got it mostly working. One final issue I think. For context this is the program right now:

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2019 Intel Corporation. All Rights Reserved.
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
// functions for rendering #include <cstring> #include <librealsense2/rs.hpp>
#include <iostream>
#include <mutex>

int main(int argc, char *argv[]) try {
  while (true) // Application still alive?
  {
    rs2::pipeline pipe;

    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
    cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F);

    pipe.start(cfg);

    rs2::frameset frameset = pipe.wait_for_frames();

    // Find and retrieve IMU data
    if (rs2::motion_frame accel_frame =
            frameset.first_or_default(RS2_STREAM_ACCEL)) {
      rs2_vector accel_sample = accel_frame.get_motion_data();
      std::cout << "Accel:" << accel_sample.x << ", " << accel_sample.y
                << accel_sample.z << std::endl;
      //...
    }

    if (rs2::motion_frame gyro_frame =
            frameset.first_or_default(RS2_STREAM_GYRO)) {
      rs2_vector gyro_sample = gyro_frame.get_motion_data();
      //   std::cout << "Gyro:" << gyro_sample.x << ", " << gyro_sample.y << ",
    }
  }
  return EXIT_SUCCESS;
} catch (const rs2::error &e) {
  //   std::cerr << "RealSense error calling " << e.get_failed_function() << "("
  //             << e.get_failed_args() << "):\n    " << e.what() << std::endl;
  return EXIT_FAILURE;
} catch (const std::exception &e) {
  //   std::cerr << e.what() << std::endl;
  return EXIT_FAILURE;
}
```
Basically, this is my output
```
Accel:0.0490332, -9.718390.755112
Assertion failed: (list_empty(&darwin_cached_devices)), function darwin_init, file os/darwin_usb.c, line 605.
[1]    80266 abort      ./rs-getting-started
```.

1. what does that even mean?
2. how should I fix it?

MartyG-RealSense, 2022-07-09:
A MacOS Monterey user experienced this error at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4478131090323-Assertion-failed-list-empty-darwin-cached-devices-function-darwin-init-file-darwin-usb-c-line-603-

At that time, there were problems with using RealSense cameras with the newly released MacOS Monterey.  A RealSense user published a detailed guide for a successful installation of the SDK on Monterey and M1.  If you are a Monterey OS user then it may be worth looking at that guide.

https://lightbuzz.com/realsense-macos/

MartyG-RealSense, 2022-07-09:
The **brew install librealsense** installation method also updated with Monterey and M1 compatibility since then.

https://formulae.brew.sh/formula/librealsense

KaustubhKhulbe, 2022-07-09:
hmmm I'm using Big Sur. and I did use brew install to download librealsense. From looking through, it seems like it's an M1 problem, and there aren't really any fixes for it?

MartyG-RealSense, 2022-07-09:
The 'brew install' compatibility table states that it should be compatible with Big Sur + M1.

[image]

Since the text-only rs-hello-realsense works on your computer, it may be worth trying the **rs-data-collect** project as it is also text-only and retrieves IMU data too.

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect

KaustubhKhulbe, 2022-07-09:
Almost got the rs-data-collect working! When I set up the config file for the D435, it works great! But I have the D435i camera, and when I c+p the config file for that, I get 
```
The following request(s) are not supported by the device: 
Type: Accel, Idx: 0, MOTION_XYZ32F, [1x1], 63fps
```
I also looked here: https://github.com/IntelRealSense/librealsense/issues/4070#issuecomment-496538805, but was unsure of what exactly to do as I never had a problem with "COLOUR".

KaustubhKhulbe, 2022-07-09:
For context, here's the config file:
```
#Video streams
DEPTH,640,480,30,Z16,0
INFRARED,640,480,30,Y8,1
INFRARED,640,480,30,Y8,2
COLOR,640,480,30,RGB8,0
#Lines starting with non-alpha character (#@!...) will be skipped
# IMU streams will produce data in m/sec^2 & rad/sec
ACCEL,1,1,63,MOTION_XYZ32F
GYRO,1,1,200,MOTION_XYZ32F
```

MartyG-RealSense, 2022-07-09:
The IMU data should be accessible on Mac, as your earlier message showed that you are using RealSense SDK version 2.50.0 and IMU data has been supported in MacOS since version 2.42.0.

I cannot see any problem in your config file.  Does it make a difference if you remove the two # comment lines above the accel definition?

KaustubhKhulbe, 2022-07-09:
Hmmm, nope that didn't do anything. You think it's an issue with me being on Big Sur and not Monterey? Also, let me know if you need any other files to better analyze.

MartyG-RealSense, 2022-07-10:
The IMU should work on Big Sur on RealSense SDK versions 2.42.0 and newer.

You could check whether your camera currently supports access to the Accel and Gyro modes without needing OpenGL support by running the text-based SDK tool **rs-enumerate-devices** - if the IMU can be accessed on your camera then Accel and Gyro should be listed at the bottom of the list of information that the tool generates when run.  If Accel and Gyro were inaccessible then they would be missing from the list.

[image]

KaustubhKhulbe, 2022-07-14:
Not sure what changed, but the same code now runs! Thanks for all your help.

MartyG-RealSense, 2022-07-14:
You are very welcome.  That's great news!

MartyG-RealSense, 2022-07-19:
Hi @KaustubhKhulbe  Do you require further assistance with this case, please?  Thanks!

KaustubhKhulbe, 2022-07-19:
Nope! I got it all sorted out. I'll close the issue.

closed 2022-07-19 by KaustubhKhulbe