realsenseai/librealsense #10667: Realtime Scan in C#

ddov123, 2022-07-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | | 
| Operating System & Version |Win 10| 
| Kernel Version (Linux Only)    | | 
| Platform                                 | PC|
| SDK Version                            |2.50.0.0| 
| Language                            |C#                          | 
| Segment			| Robot   | 

### Issue Description

Hello!

I found rs-kinfu example but It's C++ code. I need realtime scan in C# code example.
Does exist realtime scan like rs-kinfu in C# code of example?

ddov123, 2022-07-11:
Hi! @MartyG-RealSense Can I export to ply with scan a point cloud in real-time and fusion to one ply??
I apologize for my bad English. Thank you.

ddov123, 2022-07-12:
Wow! @MartyG-RealSense Thank you!!
and.... back to rs-kinfu, 
[image]
this error occurred.

Help me.

MartyG-RealSense, 2022-07-12:
A RealSense user who experienced this problem with rs-kinfu on Windows at https://github.com/IntelRealSense/librealsense/issues/9432 solved it by adding **msvcrtd.lib** to the dependencies.

ddov123, 2022-07-13:
@MartyG-RealSense It's not working... 
Well, Do I need cuda when using rs-kinfu??

MartyG-RealSense, 2022-07-13:
No, CUDA is not a requirement for rs-kinfu.

ddov123, 2022-07-13:
@MartyG-RealSense Yes, I used
OpenCV 4.5.0
OpenCV_contrib 4.5.0
glfw 3.3
and in console, I found this

`OpenCL error CL_OUT_OF_RESOURCES (-5) during call: clEnqueueNDRangeKernel('fillPtsNrm', dims=3, globalsize=512x512x512, localsize=8x8x16) sync=true`

MartyG-RealSense, 2022-07-13:
This error has occurred before with kinfu, as described at https://github.com/opencv/opencv_contrib/issues/2029

Kinfu was not originally designed for the RealSense SDK, and rs-kinfu is an adaptation of Kinfu for RealSense cameras.  Within the case linked to above, a RealSense user reported at https://github.com/opencv/opencv_contrib/issues/2029#issuecomment-477004241 that they were experiencing the same error with rs-kinfu.  That user experienced errors with both an Intel integrated graphics GPU and with an Nvidia GPU.  They found that the errors disappeared if OpenCL optimization was disabled using a method described in their comment.

ddov123, 2022-07-14:
It stopped for a while, then error again...
I think 

In 'colorize_pointcloud'
`std::vector<Mat> channels(points.channels());` 
`split(points, channels);` 
`color = channels[2];`
here vector error occur.

Thank you. I'll looking for more.

MartyG-RealSense, 2022-07-14:
Thanks very much @ddov123 for the update!

ddov123, 2022-07-14:
@MartyG-RealSense OMG!!!!! I solved it!!
I don't know why solved it, just I've build 5 more times.
build , again build, again build...
Thank you very much! Hero.

MartyG-RealSense, 2022-07-14:
It's excellent to hear that you were successful, @ddov123   :)

closed 2022-07-14 by ddov123