realsenseai/librealsense #10668: In C# How to get the depth of the (x, y) coordinates from depth data ?

SeokHoYa, 2022-07-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.04.01 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0.0                          | 
| Language                            |  C#                         | 
| Segment			|  others                         | 

Hello
First, I am sorry that I'm not familiar with English.
I have purchased 1 units of L515.
My goal is to measure the depth of the (x, y) coordinates (not using GetDistance function).
In C# I used the following code to get the depth of the (x, y) coordinates where the frame has dimensions of (320x240) depthFrame. But it does not work.

a. code
float objectDistance01 = depthFrame.GetDistance(depthFrame.Width / 2, depthFrame.Height / 2);
///objectDistance01 = 0.724750042
var depthData = new ushort[depthFrame.Width * depthFrame.Height];
depthFrame.CopyTo(depthData);
int x = depthFrame.Width / 2;
int y = depthFrame.Height / 2;
float objectDistance02 = depthData[(y * depthFrame.Width + x)] / 1000;
///objectDistance02 = 2

b. reference code
https://github.com/IntelRealSense/librealsense/issues/10393
var depthData = new ushort[depthFrame.Width * depthFrame.Height];
depthFrame.CopyTo(depthData);
DepthDataStruct _depthdata;
_depthdata.depthData = depthData;
_depthdata.rows = depthFrame.Height;
_depthdata.cols = depthFrame.Width;
_ds.depthData[(y * _ds.cols + x)]/1000;

How to get the depth of the (x, y) coordinates ?
Thanks in advance.

MartyG-RealSense, 2022-07-12:
Hi @SeokHoYa   https://github.com/IntelRealSense/librealsense/issues/10393#issuecomment-1097519036 states that the result of the method used in the comment is the same as using GetDistance.  A simpler approach may therefore be to use the GetDistance code for C# at https://github.com/IntelRealSense/librealsense/issues/4253#issuecomment-504964580 that uses **deprotection** to find the Z depth value.

SeokHoYa, 2022-07-12:
Thank you for your reply.

closed 2022-07-12 by SeokHoYa