realsenseai/librealsense #10670: depth frame getdata()just get all zero

guyuezuntinggithub, 2022-07-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info
|---------------------------------|------------------------------------------- |
| Camera Model | {D435i } |
| Firmware Version | (05.11.06.200) |
| Operating System & Version | win10 |

| Platform | PC |
| SDK Version | { 2.21.0 } |
| Language | {C++ } |
| Segment | {Robot} |

### Issue Description
<Describe your issue / question / feature request / etc..>
here is part code
frames = pipe.wait_for_frames();
			if (frames) {
				
				//postProcessing(frames);
				/*int num_of_frames = rs2_embedded_frames_count(frames, &e);*/
				// align images
				frames = depthToColorAlignment.process(frames);

				// Let's get our depth frame
				rs2::depth_frame depth = frames.get_depth_frame();

I check every single number by (uint16_t *)depth.get_data().and get all number is zero! there is nothing! but the following is work well and can get point cloud
 rs2::pointcloud pc;
rs2::points points = pc.calculate(depth);

why!!!!!!!!!!!

MartyG-RealSense, 2022-07-12:
Hi @guyuezuntinggithub  Are you able to obtain depth with the C++ code at https://github.com/IntelRealSense/librealsense/issues/4026#issuecomment-496162368 please?

guyuezuntinggithub, 2022-07-13:
> Hi @guyuezuntinggithub Are you able to obtain depth with the C++ code at [#4026 (comment)](https://github.com/IntelRealSense/librealsense/issues/4026#issuecomment-496162368) please?

> Hi @guyuezuntinggithub Are you able to obtain depth with the C++ code at [#4026 (comment)](https://github.com/IntelRealSense/librealsense/issues/4026#issuecomment-496162368) please?

very appreciate for your reply!!! I will try it soon.

guyuezuntinggithub, 2022-07-18:
I still can't get data ,but use Z and depth scale get depth data

closed 2022-07-18 by guyuezuntinggithub