realsenseai/librealsense #10674: Realsense Camera experimental conditions as radioactive environnement

pltrd, 2022-07-13:
Hi everybody, 
I am working on a realsense camera D455. I need to put my camera in a radioactive environnement and make signal treatment on the pointcloud provided by the camera using SDK and post-process filters
I have not any notions on how a point cloud can be disturbed by nuclear environnement and about radioactive environnement in general. Has anyone worked on something like this before? If yes, do you have any recommendations or suggestions ? Maybe it is possible to make a calibration of the pointcloud according to a certain value of radioactivity ?
Best regards:)

MartyG-RealSense, 2022-07-13:
Hi @pltrd   Whilst I have not seen a past practical example of such a project with a RealSense 400 Series camera, in theory I cannot foresee a situation where invisible radiation would have a disruptive effect on a depth pointcloud unless the radiation was producing light wavelengths within the range of the camera's perception (from 400 nm to 1000 nm).

MartyG-RealSense, 2022-07-20:
Hi @pltrd  Do you require further assistance with this case, please?  Thanks!

pltrd, 2022-07-21:
Hi MartyG,
Sorry for responding late, for the moment, I got what I need, I'll be come
back to you if needed.
Thank you

MartyG-RealSense, 2022-07-21:
You are very welcome, @pltrd - good luck!

MartyG-RealSense, 2022-08-01:
Case closed due to no further comments received.  Please feel free to re-open this case or create a new one if you need to do so at a future date.  Thanks!

closed 2022-08-01 by MartyG-RealSense