realsenseai/librealsense #10675: Depth map quality

L-xn, 2022-07-14:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455| 
| Operating System & Version |   {Win 10  | 
| Language                            | python }                          | 
Sorry to bother you again!I'm now studying how mice behave at night.The first thing to do is track the mice from a depth map.But I found that when the mouse stood up against the wall, part of its body formed the same color as the wall.This made it difficult for me to track the full outline of the mouse.The first chart below is easy to track.But in the second image, it's harder to get an accurate outline of the mouse.
[image]
[image]

MartyG-RealSense, 2022-07-14:
If some black tape was put around the edge of the bucket at the bottom then that may create a clearer separation, as depth cameras have difficulty with reading dark grey or black color shades because they absorb light.  Those color areas can therefore appear on the image as plain black areas without depth information.  When the mouse stands on that tape at the edge they should therefore stand out better on the depth image, especially if the tape extends part-way up the side of the base of the bucket as well as round its bottom-most edge.

L-xn, 2022-07-16:
Thank you for your reply！That's a good way to do it!But I don't have any mice to film right now. I can only process these videos that I've shot before.Obviously, the problem of cylinder walls cannot be solved by background subtraction.I have tried to use UNET neural network and Labelme to train marked pictures these days.Now I have an idea, can it be better to divide mice through point cloud?How to extract point cloud from BAG file in Python?

MartyG-RealSense, 2022-07-16:
You could try different colorization and color scheme options with your bag in the Viewer's **Depth Visuaization** controls (which work with bag files) and then program the changes into your Python application if they make the mouse stand out against the wall more clearly.

[image]

References for altering the color settings in Python can be found at https://github.com/IntelRealSense/librealsense/issues/7767#issuecomment-726704671 and https://github.com/IntelRealSense/librealsense/issues/7089#issuecomment-672105974

You could generate a point cloud from a bag in the SDK's **opencv_viewer_example** program by using a live-camera pointcloud script and adapting it to use the bag as its data source instead of the camera by placing a **enable_device_from_file** instruction on the line before the pipe.start instruction, as demonstrated in the SDK's **read_bag_example.py** Python example program.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py#L42

MartyG-RealSense, 2022-08-01:
Hi @L-xn  Do you still require assistance with improving depth map quality or can this particular part of your ongoing mouse behaviour case be closed, please?  Thanks!

L-xn, 2022-08-01:
I really appreciate your help!

MartyG-RealSense, 2022-08-07:
Hi @L-xn  Is the mouse still difficult to see separately on the depth image when it is at the side of the bucket?

MartyG-RealSense, 2022-08-14:
Hi @L-xn  Can you confirm please if this case can now be closed or if you need more help?  Thanks!

closed 2022-08-20 by MartyG-RealSense