realsenseai/librealsense #10676: Install for MacOS Monterey

mingyangpeng, 2022-07-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                       Install for MacOS Monterey                                         |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (L515  newest 1.5.8) | 
| Operating System & Version |   {MacOS Monterey 12.4 -- M1 PRO | 
| Platform                                 | MBP  |
| SDK Version                            |  { legacy / 2.5 }                          | 
| Language                            |  {C++ / python}                          | 
| Segment			|  {others }                          | 

### Issue Description
Is there a simple installation method for MacOS monterey M1 pro? I tried “ brew install librealsense”, but error ：failed to set power state，I read the previous installation method and tried it, but the errors were the same， I need help......

MartyG-RealSense, 2022-07-15:
Hi @mingyangpeng  A RealSense user created and shared a detailed RealSense installation guide for Monterey and M1 at https://lightbuzz.com/realsense-macos/

The **brew install librealsense** installation method lists support for Monterey and M1.  Could you test please whether the 'Failed to set power state' error disappears if you unplug the micro-sized end of the USB cable from the base of the L515 camera, turn the connector around the other way and re-insert it into the base of the camera (USB Type-C cables are two way insertion at the micro-sized end).

https://formulae.brew.sh/formula/librealsense

mingyangpeng, 2022-07-15:
Hi, @MartyG-RealSense ，thank you for your advices, I try to insert USB Type-C in both directions, and the error information is as follows：

`sudo realsense-viewer`

plug with face up :
```
> 15/07 16:26:56,657 WARNING [0x16fe6b000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Operation timed out, number: 3c
>  15/07 16:27:01,873 ERROR [0x16fe6b000] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
>  15/07 16:27:01,873 ERROR [0x17009b000] (sensor.cpp:572) acquire_power failed: failed to set power state
>  15/07 16:27:01,873 ERROR [0x17009b000] (error-handling.cpp:78) Error during polling error handler: failed to set power state
>  15/07 16:27:01,994 WARNING [0x16fe6b000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Operation timed out, number: 3c
>  15/07 16:27:02,109 WARNING [0x16fe6b000] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Operation timed out, number: 3c
```

plug with face down :
```
> 15/07 16:31:01,705 WARNING [0x16fb47000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
>  15/07 16:31:06,921 ERROR [0x16fb47000] (handle-libusb.h:100) failed to claim usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
>  15/07 16:31:06,921 ERROR [0x16fd77000] (sensor.cpp:572) acquire_power failed: failed to set power state
>  15/07 16:31:06,921 ERROR [0x16fd77000] (error-handling.cpp:78) Error during polling error handler: failed to set power state
```

I also tried this way “https://lightbuzz.com/realsense-macos/”. I follow the steps in the article, I'm not see the RealSense target under the Products folder,I didn't find librealsense2.2.50.0.dylib , I'm not familiar with Xcode, I gave up.

Thank you again for your advice，Is there any other simple installation method？

MartyG-RealSense, 2022-07-15:
For the M1 Macs, brew or the Lightbuzz guide are currently the best methods of performing an installation of the SDK.

It sounds as though the brew install is functional, although you are receiving a power state error.  What RealSense program is producing the error, please?

mingyangpeng, 2022-07-16:
Sorry to reply you so late, These errors are produced by `sudo realsense-viewer`, I hope you can give a solution that can work normally based on MacOS Monterey.

MartyG-RealSense, 2022-07-16:
Other RealSense Mac users have experienced this issue with launching realsense-viewer with sudo, an example of which is https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-1048069662

Since that comment was published, the previously mentioned Lightbuzz installation guide for Monterey / M1 and support for them in 'brew install librealsense' have been released.  However, you are still experiencing the problem.

The best place to ask for advice on this subject is likely to be the main Monterey / M1 discussion thread at https://github.com/IntelRealSense/librealsense/issues/9916 where fellow Mac M1 users are very willing to offer advice to other users who still have problems.

mingyangpeng, 2022-07-17:
Hi,@MartyG-RealSense, I've seen #9916 before, but it doesn't work. So far from last year, there is no definite solution to the USB connection problem. I hope your official team can adapt and reconstruct it on MacOS Monterey.

MartyG-RealSense, 2022-07-17:
Hi @mingyangpeng  MacOS development work for the RealSense SDK ended after MacOS High Sierra and although the SDK has continued to be compatible with subsequent MacOS versions such as Catalina and Big Sur (and now Monterey), there are no plans for further Mac development by the Intel RealSense development team unfortunately.

mingyangpeng, 2022-07-17:
This is really bad news,, which will hinder the development of the RealSense，If there is any new development in the later stage, please let me know. Thank you！

MartyG-RealSense, 2022-07-17:
Thanks very much for your understanding.  Mac members of the RealSense community can continue to submit updates that they have created via a Pull Request so that they can be reviewed by the RealSense development team for merging into the RealSense SDK.

https://github.com/IntelRealSense/librealsense/pulls

MartyG-RealSense, 2022-07-23:
Hi @mingyangpeng  Do you require further assistance with this case, please?  Thanks!

mingyangpeng, 2022-07-25:
Hi,@MartyG-RealSense I hope to get further help. Let's keep this, Thank you!

MartyG-RealSense, 2022-07-25:
Okay, it is no problem to keep this open for other members of the RealSense Mac community to comment on.  Thanks!

mingyangpeng, 2022-07-25:
@MartyG-RealSense, thank you for your help!

MartyG-RealSense, 2022-11-19:
Case closed due to no further comments after 4 months.

closed 2022-11-19 by MartyG-RealSense