realsenseai/librealsense #10678: Background segmentation noise

DutchRose-Jochem, 2022-07-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |  Win 11  | 
| Platform                                 | PC  |
| SDK Version                            | 2.50.0                           | 
| Language                            | unity                           | 
| Segment			|  Background segmentation                          | 

### Issue Description
I keep getting noise and seeing holes when trying to use background segmentation in Unity, with the D455 (see screenshot). The same happens in the viewer. I think I tried every setting, but nothing seems to really help. Am I doing something wrong? Or is the RealSense just not really meant for seamless segmentation?

[image]

MartyG-RealSense, 2022-07-15:
Hi @DutchRose-Jochem  As you are using the D455 camera model that has an ideal depth sensing distance of 0.6 m to 6 m, the detail of the body areas nearest the camera (such as the face leaning forwards) may be being excluded from the image.  

You could confirm whether this is the case by standing a little further away from the camera.  Alternatively, you could try reducing the minimum depth sensin distance in the RealSense Viewer by going to **Stereo Module > Advanced Controls > Depth Table** and setting the 'Disparity Shft' option to **100** instead of the default '0'.  This will reduce the camera's minimum distance (though also reduce the maximum observable distance) and enable objects / surfaces to be closer to the camera before the depth detail starts progressively breaking up as the observed surface moves further below the minimum distance.

[image]

DutchRose-Jochem, 2022-07-15:
Hi @MartyG-RealSense ,

Thanks for the quick response!
Yeah, the screenshot maybe wasn't the best example. I've attached a screen recording giving a better view of the issue I'm facing. It's still very jittery and not as smooth as we would like.

https://user-images.githubusercontent.com/40468550/179238071-43228f4c-51aa-4f4d-9a37-075abe39a783.mp4

MartyG-RealSense, 2022-07-15:
I tested the **BackgroundSegmentation** example and noted that it was set by default to 640x480 resolution for both depth and color.  Do you have better performance if you select the **RsDevice** object and set both depth and color to 1280x720 before starting the program with the play button?

DutchRose-Jochem, 2022-07-21:
Yes, that's one of the first things I tried. The screen recording has both the depth and color on 1280x720.

MartyG-RealSense, 2022-07-21:
I ran a further set of tests.  I found that the loss of detail on the face will be greatest if:

(a) Feather is set to '0' like it is in your original image above.  Any value above '0' restored missing detail. and 

(b) if the camera is too close to the human.  Missing detail on the face will return if the camera is moved slightly further away. and

(c) the eyes will tend to disappear if BG Opacity is set to the level that you have it at in the above image and the face is too close to the camera.

MartyG-RealSense, 2022-07-27:
Hi @DutchRose-Jochem  Do you have an update about this case that you can provide, please?  Thanks!

DutchRose-Jochem, 2022-08-01:
Hi @MartyG-RealSense, sorry. I have Covid at the moment, so I haven't had the chance to check out the things you've mentioned. I really appreciate the help and will let you know as soon as I've managed to try your suggestions.

MartyG-RealSense, 2022-08-01:
Okay, thanks very much for the update.  I wish you a quick recovery!

DutchRose-Jochem, 2022-08-09:
Hi, I'm back at work again and tried your suggestions, but didn't manage to get much better results, unfortunately.

MartyG-RealSense, 2022-08-10:
Is use of Unity for your RealSense segmentation project compulsory, please?

The RealSense SDK's C++ and OpenCV example program**rs-grabcuts** provides more solid segmentation than the image that you have been experiencing in the Unity segmentation example.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/grabcuts

DutchRose-Jochem, 2022-08-12:
I'm gonna look into that, thanks!

DutchRose-Jochem, 2022-08-24:
Hi @MartyG-RealSense, of course!

There have been some developments in the scope of the project, making it impossible for us to step away from Unity for now. We came across a sample project using Barracuda to remove the background image from just a regular webcam. This quality seems to be good enough.

I will still try out your OpenCV suggestion, as soon as I have time. So maybe we can still use it for a different project.

MartyG-RealSense, 2022-08-24:
Thanks very much for the update!  Let's keep this case open for a while longer, and if you decide to close it then you are welcome to do so using the **Close Issue** button under the comment writing box.  Good luck!

MartyG-RealSense, 2022-10-21:
Case closed due to no further comments received after 2 months.

closed 2022-10-21 by MartyG-RealSense