realsenseai/librealsense #10681: Issue with intel realsense viewer

aryasuryapp, 2022-07-18:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2022-07-18:
Hi @aryasuryapp  Could you try downloading the standalone version of the Viewer called **Intel.RealSense.Viewer.exe** from the 'Assets' file list on the SDK Releases page if you are not already using this Viewer version and see whether the error still occurs, please?  

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

[image]

The standalone Viewer should be launched with a double-leftclick from the folder that it is downloaded to, and not with the Viewer launch icon on the Windows desktop.

aryasuryapp, 2022-07-18:
it happened when I try to stream from my laptop built in camera

MartyG-RealSense, 2022-07-18:
Do you select the laptop's RGB camera using the **Add Source** button at the top of the Viewer's options side-panel please?

[image]

aryasuryapp, 2022-07-18:
yes, i done it and the error pop up

MartyG-RealSense, 2022-07-24:
Hi @aryasuryapp  Do you require further assistance with this case, please?  Thanks!

aryasuryapp, 2022-07-28:
Yeah, please help me, i need to use this for college final project

MartyG-RealSense, 2022-07-28:
Can you confirm that you are not changing any of the controls in the Viewer's options side-panel before you begin streaming.  You select the Platform Camera and then enable the RGB Camera stream and get the **Unhandled exception escaping from a worker thread** error.

[image]

aryasuryapp, 2022-07-28:
Yes, that is correct.

MartyG-RealSense, 2022-07-28:
I would recommend avoiding making changes to the laptop's webcam driver in the same way that uninstalling and reinstalling the RealSense camera drivers is done to correct a problem, as it could stop the webcam from working at all outside of RealSense since you may not be able to reinstall the built-in cam's driver once it was removed.

Would it be possible to purchase a cheap miniature external webcam that can clip onto the top of the laptop screen and see whether you can use that as the platform camera instead (google for the term 'miniature clip on webcam' for examples of such cameras).

MartyG-RealSense, 2022-07-28:
If you did want to try uninstalling and reinstalling the built-in camera's driver (completely at your own risk), you could do so by:

1.  Going to the 'Camera' section of the Windows Device Manager, right-clicking on the driver for the built-in laptop camera (which might be called "Integrated Camera" and selecting the 'Uninstall device' menu option.

[image]

2.  After the driver has been removed from the Device Manager, left-click on the 'Scan for hardware changes' option on the Device Manager.  

[image]

It should then check for changes and re-install the built-in camera driver into the 'Cameras' section of the Device Manager.

MartyG-RealSense, 2022-07-28:
Before changing the built-in camera driver though, please first check whether permission to access the camera hardware and permission for applications to access the camera are both set to 'On' in the Windows camera privacy settings.  This can be done using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

closed 2022-08-10 by MartyG-RealSense