realsenseai/librealsense #10682: D405 camera software for 3dscan and python examples not working

jeojeda, 2022-07-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Win 11)  | 
| Platform                                 | PC |
| SDK Version                            |  SDK-WIN10-2.50.0.3785                          | 
| Language                            |  python  | 
| Segment			| Robot | 

Hello, I have not been able to use the D405 camera with the python examples since an error is generated indicating "The demo requires Depth camera with Color sensor".

I also haven't been able to use the software that Intel recommends for 3d scanning using a turntable, such as LIPSCAN 3D or Dot3D. I would like you to help me with the type of software or codes that can be used since we acquired this camera thinking that we could make measurements and reconstructions of the deformation suffered by metal sheets due to the effect of temperature and then send the data to a welding robot.
Do you have any example on how to run the SDK from scratch?

MartyG-RealSense, 2022-07-18:
Hi @jeojeda   The D405 has a space-saving hardware design that provides RGB color from the depth sensor instead of having a separate RGB sensor component like RealSense models such as D415, D435 / D435F / D435i and D455 do.

RealSense-compatible software tools produced by companies other than Intel may need to be updated by those companies to support the D405 camera model specifically.  

The D405 is fully compatible with tools and examples bundled with the RealSense SDK.  On Windows, if you have installed the full SDK with the **Intel.RealSense.SDK-WIN10** installer program then you can find these compatible tools and examples by right-clicking on the RealSense Viewer launch shortcut icon on the Windows desktop and selecting the **Open file location** menu option to open the SDK's 'Tools' folder.

MartyG-RealSense, 2022-07-26:
Hi @jeojeda  Do you require further assistance with this case, please?  Thanks!

jeojeda, 2022-07-27:
Hello @MartyG-RealSense. 

I am using the examples codes with the adjustments you proposed in #10445. I am still trying to get a result of an object scanning but without any third party software. This brings me to the next point. The point clouds I get from the parts I am capturing have a lot of noise and wavy, so the dimension records I can then make are not accurate as the points change position as I take different shots. I am trying to get millimeter detail on an 11 x 20 cm piece.

Is there any calibration parameter to adjust that will allow me to get a kind of "static" capture of a fixed object within the video? This is because the goal is to make a constant measurement of an object that is being affected by temperature, so I want to record the deformation of some points on the surface.

MartyG-RealSense, 2022-07-27:
If you are experiencing depth fluctuations then applying a **temporal filter** and setting its 'Filter Smooth Alpha' value to 0.1 instead of its default of 0.4 can significantly stabilize fluctuation.  

You can test this in the RealSense Viewer by going to **Post-Processing > Temporal Filter**, expanding open the Temporal filter's settings and setting the Alpha to 0.1.  You can enter the value manually with the keyboard instead of using the slider by left-clicking on the pencil icon at the end of the setting.

[image]

https://github.com/IntelRealSense/librealsense/issues/10078#issuecomment-997292916 has Python code for setting up the temporal filter and its alpha value.

[image]

jeojeda, 2022-08-02:
hi @MartyG-RealSense.
Yes, I have some more questions.

Right now I have this error:

Traceback (most recent call last):
  File "c:\IntelRealSense\vs_realsense\realsense_auto_canny.py", line 5, in <module>
    from realsense_depth import *
ModuleNotFoundError: No module named 'realsense_depth'
PS C:\IntelRealSense\vs_realsense> 

I am importing this:
import numpy as np
import cv2
import pyrealsense2
from pytest import PytestUnknownMarkWarning
from realsense_depth import *

jeojeda, 2022-08-02:
Hi @MartyG-RealSense. Yes, I still have some questions.
This is an example of what I would like to measure. In the picture you can see a piece of clay that is not flat and the deformed parts or "errors" on the surface are the ones I want to measure and reconstruct so I can use those points as a reference for a robot movement.
For now I am trying with some captures and opencv to get detections of corners or points to extract the position of the point in space but the result it is not fixed if I use the video.

[image]
[image]
[image]
[image]
[image]

Do you have any suggestions on how to configure the camera parameters, and if it is possible to load this configuration in the same script with opencv?

MartyG-RealSense, 2022-08-02:
The images above are from the **Depth Quality Tool** which provides feedback about the current depth accuracy of the camera.   The better tool to use for this particular activity is likely to be the RealSense Viewer tool.  

If you launch the Viewer and left-click on the '3D' option in the top corner of the Viewer's window to enter 3D pointcloud mode then you can see the XYZ real-world coordinates of a specific 3D point on the image displayed as a text overlay by moving the mouse cursor over it.

If you wanted to reduce depth fluctuation with a setting that was supported in a camera configuration file then you could try setting the Advanced Mode option **Second Peak Threshold' to '0' instead of its default '325' and then export the Viewer's settings as a **json** using the json save option next to the Visual Preset menu selector near the top of the Viewer side-panel.  

The export will save the defined Second Peak Threshold value into the json as the setting **param-secondpeakdslta**, as described at https://github.com/IntelRealSense/librealsense/issues/4993#issuecomment-550184812

A json can be loaded into a C++ or Python script to apply the settings stored in the file.

jeojeda, 2022-08-05:
Thank you @MartyG-RealSense 
Any idea why the focal length is not working?
The on-chip calibration works because I can measure an object afterwards and the mm response is good. Do I have to calibrate every time I use the camera? When I use it again the measurement changes.
I can't calibrate the focal length. I always have an error. is there a good setting to calibrate, such as white paper on the wall, natural light, color of the object, etc.?
[image]
[image]

MartyG-RealSense, 2022-08-05:
The focal length calibration requires a calibration target image specially designed for focal length (it can be used with Tare depth accuracy calibration too).  The image would be printed out from a PDF document and the printout attached to the wall.

You can obtain PDFS of the focal length target in two paper sizes at the link below.
https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#ground-truth-targets-and-printing--mounting-instructions

jeojeda, 2022-08-09:
Thanks @MartyG-RealSense
I am now getting better results with the parameters you mentioned. Can you please check if it would be ok to change the Maximum Score Threshold or adjust another parameter that would allow me to achieve this result? 

[image]<p>
[image]<p>
[image]<p>

With this parameter set to 0 instead of 2047 the noise is reduced, or at least in the preview I don't get waves. The only change I have is that the depth feedback does not work and becomes static (if I move the pen out of the frame the depth view does not update).

[image]<p>
[image]<p>

MartyG-RealSense, 2022-08-09:
Most of the Advanced Mode options are not documented.  This is because they interact with each other in complex ways and so Intel chose to control them with machine-learning algorithms.  RealSense users are free to experiment with the settings to see how they affect the image.  If you require advice about one specific parameter such as **Score Maximum Threshold** then I can consult my Intel RealSense colleagues.

With Score Maximum Threshold set to minimum, you could try changing the Temporal filter's alpha from 0.1 back to its default of 0.4 or a higher value to see whether the depth frame updates after motion again.  This is because reducing the alpha to 0.1 reduces depth fluctuation by decreasing how quickly the image updates.

jeojeda, 2022-08-09:
I am getting very good results with statics point clouds capture. 
[image][image][image]<p>

I would like to get the same results but with a non-static version inside opencv instead of a point cloud that for now I can only analyze with a linear workflow using [CloudCompare](https://www.danielgm.net/cc/). I think with the image and the depth camera value it could be easier. The final goal is to have feedback of the deformation by selecting some points on a grid painted or printed on the part.
Do you have any suggestions? Thank you very much for your help.

[image]<p>
[image]<p>

MartyG-RealSense, 2022-08-09:
Normally you could get a sense of the curvature of an object with a 400 Series camera by projecting an infrared dot pattern onto the surfaces of an object, but the D405 does not have a projector and also cannot see infrared frequencies.  You can though use an external **visible pattern* projection that the D405 will be able to see and include in an RGB image.  Details of using a visible pattern projection with a RealSense 400 Series camera are in Intel's white-paper guide about projectors at the link below.

https://dev.intelrealsense.com/docs/projectors#6-the-consumer-front-projector

A projected visible pattern should also add texture to surfaces, making it easier for the camera to analyze them for depth information.

********

A more advanced way to analyze a part for deformation would be to scan the part and compare it to an ideal flawless CAD model, like in the discussion at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360038418393/comments/360010005694

MartyG-RealSense, 2022-08-21:
Case closed due to no further comments received.

closed 2022-08-21 by MartyG-RealSense

jeojeda, 2022-09-02:
Hi @MartyG-RealSense, sorry for the delay in replying. I was out of the office.
I have this code, and maybe it's not clean and pretty but so far it works to detect the points I want from a pattern and then I can export them to Grasshopper. The problem is that I am exporting only the Z coordinate as a real distance, but now the x and y coordinates are the pixels. Is there any code to get the coordinates that I can see in the RealSense Viewer?

```
import string
from flask import Flask
import ghhops_server as hs
import rhino3dm

from ssl import PROTOCOL_TLSv1_2
import pyrealsense2 as rs
import numpy as np
import cv2 as cv2

########################
####          HOPS              ####
########################
# Hops register hops app as middleware
app = Flask(__name__)
hops: hs.HopsFlask = hs.Hops(app)
@hops.component(
    "/test",
    outputs=[
    hs.HopsString("x coord", "x", "x coordinates"),
    hs.HopsString("y coord", "y", "y coordinates"),
    hs.HopsString("z coord", "z", "z coordinates")],
)
def SendText():
    return to_ghx, to_ghy, to_ghz
def runHops():    
    # if __name__ == "__main__":
    #     app.run(debug=True)
    app.run(debug=True)

##############################
##              Realsense code               ##
##############################
# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()

# Get device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30) #640,480
config.enable_stream(rs.stream.color, 848, 480, rs.format.bgr8, 30) #640,480

# Start streaming
pipeline.start(config)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        depth_colormap_dim = depth_colormap.shape
[... 71 lines cut ...]
                global to_ghx
                to_ghx = (str(coord_x)) 
                global to_ghy
                to_ghy = (str(coord_y))
                global to_ghz
                to_ghz = (str(coord_z))

                cv2.imshow('RealSense', images)
                cv2.imshow('Contours By Area', blurred)
    
        
        if cv2.waitKey(0) == 27 or cv2.waitKey(0) == ord('q'):
            cv2.destroyAllWindows() 
            break
       

finally:
    
    # Stop streaming
    pipeline.stop()
    cv2.destroyAllWindows() 
    runHops()
    
    
```

MartyG-RealSense, 2022-09-02:
Hi @jeojeda  If you wanted to obtain real-world 3D XYZ coordinates (not just Z-depth) with Python and you are storing the depth and color data in numpy arrays (as your script does) then I would normally recommend:

1.  Perform depth to color alignment.

2.  **Deproject** the data with the SDK instruction **rs2_deproject_pixel_to_point** to obtain XYZ real-world coordinates.