realsenseai/librealsense #10683: Align process error when reading from .bag (Python, Win 10, SDK 2.50.0)

mh60k, 2022-07-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Win 10   | 
| Platform                                 | PC  |
| SDK Version                            |   2.50.0                           | 
| Language                            |  python                        | 
| Segment			|  others                          | 

### Issue Description
Hi, I wrote a little python code to use a L515 for scientific research. I recorded to a bag file and use that file to run the program. When I wrote it, in May 2022, it worked well. When I tried it last week, I got an error code when it tried to run the "aligned_frames = align.process(frames)" line of the code below.

```python
import pyrealsense2 as rs
import numpy as np
import math

pipe = rs.pipeline()

cfg = rs.config()
cfg.enable_device_from_file("filepath.bag")
cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
profile = pipe.start(cfg)

align_to = rs.stream.depth
align = rs.align(align_to)
        
frames = pipe.wait_for_frames()

#Aligning color frame to depth frame
aligned_frames =  align.process(frames)
depth_frame = aligned_frames.get_depth_frame()
aligned_color_frame = aligned_frames.get_color_frame()

color_intrin = aligned_color_frame.profile.as_video_stream_profile().intrinsics
depth_image = np.asanyarray(depth_frame.get_data())
color_image = np.asanyarray(aligned_color_frame.get_data())
#Use pixel value of  depth-aligned color image to get 3D axes
x, y = 360, 360
depth = depth_frame.get_distance(x, y)
dx ,dy, dz = rs.rs2_deproject_pixel_to_point(color_intrin, [x,y], depth)
distance = math.sqrt(((dx)**2) + ((dy)**2) + ((dz)**2))
print("Distance from camera to pixel:", distance)
print("Z-depth from camera surface to pixel surface:", depth)
```

Any idea what might be wrong?

Thank you!

mh60k, 2022-07-18:
Hi Marty, when I add that third line, I get another error reading "RuntimeError: Couldn't resolve requests" when starting the pipeline.

MartyG-RealSense, 2022-07-18:
The color line should match the resolution and FPS of the color stream stored in the bag file.  If there is no color data within the bag file then the script may not be able to access color.  The **RuntimeError: Couldn't resolve requests** error means that either there is a typing error in the instruction (which there isn't in this case) or the requested configuration could not be provided.

mh60k, 2022-07-18:
Thank you so much Marty! I managed to make it work with the proper resolution info. Sorry to have bothered you with this!

MartyG-RealSense, 2022-07-18:
It's no bother at all.  :)  It's great to hear that you were successful

closed 2022-07-26 by MartyG-RealSense